chenmozhijin/turboacc #6: 编译时发生错误

xiamo23, 2023-07-09:
CC [M]  net/netfilter/nf_conntrack_proto_tcp.o
In file included from ./include/linux/linkage.h:7,
                 from ./include/linux/kernel.h:8,
                 from ./include/linux/list.h:9,
                 from ./include/linux/timer.h:5,
                 from net/netfilter/nf_conntrack_proto_tcp.c:9:
net/netfilter/nf_conntrack_proto_tcp.c:35:19: error: 'nf_ct_tcp_no_window_check' undeclared here (not in a function)
   35 | EXPORT_SYMBOL_GPL(nf_ct_tcp_no_window_check);
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~
./include/linux/export.h:104:23: note: in definition of macro '___EXPORT_SYMBOL'
  104 |         extern typeof(sym) sym;                                                 \
      |                       ^~~
./include/linux/export.h:161:41: note: in expansion of macro '__EXPORT_SYMBOL'
  161 | #define _EXPORT_SYMBOL(sym, sec)        __EXPORT_SYMBOL(sym, sec, "")
      |                                         ^~~~~~~~~~~~~~~
./include/linux/export.h:165:41: note: in expansion of macro '_EXPORT_SYMBOL'
  165 | #define EXPORT_SYMBOL_GPL(sym)          _EXPORT_SYMBOL(sym, "_gpl")
      |                                         ^~~~~~~~~~~~~~
net/netfilter/nf_conntrack_proto_tcp.c:35:1: note: in expansion of macro 'EXPORT_SYMBOL_GPL'
   35 | EXPORT_SYMBOL_GPL(nf_ct_tcp_no_window_check);

请问一下这是什么情况?吗，同时编译了MWAN3,WIREGUARD,UPNP

chenmozhijin, 2023-07-09:
什么分支？是按照使用方法执行的吗？

xiamo23, 2023-07-09:
> 

 删除SFE后可以正常编译了！估计可能是我哪步没弄好，SFE不是刚需 就没重新弄了！谢谢答复

skullcrossbones, 2023-07-16:
net/netfilter/nf_conntrack_proto_tcp.c:35:19: error: 'nf_ct_tcp_no_window_check' undeclared here (not in a function)
   35 | EXPORT_SYMBOL_GPL(nf_ct_tcp_no_window_check);
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~
./include/linux/export.h:104:23: note: in definition of macro '___EXPORT_SYMBOL'
  104 |         extern typeof(sym) sym;                                                 \
      |                       ^~~
./include/linux/export.h:161:41: note: in expansion of macro '__EXPORT_SYMBOL'
  161 | #define _EXPORT_SYMBOL(sym, sec)        __EXPORT_SYMBOL(sym, sec, "")
      |                                         ^~~~~~~~~~~~~~~
./include/linux/export.h:165:41: note: in expansion of macro '_EXPORT_SYMBOL'
  165 | #define EXPORT_SYMBOL_GPL(sym)          _EXPORT_SYMBOL(sym, "_gpl")
      |                                         ^~~~~~~~~~~~~~
net/netfilter/nf_conntrack_proto_tcp.c:35:1: note: in expansion of macro 'EXPORT_SYMBOL_GPL'
   35 | EXPORT_SYMBOL_GPL(nf_ct_tcp_no_window_check);
      | ^~~~~~~~~~~~~~~~~
make[7]: *** [scripts/Makefile.build:286: net/netfilter/nf_conntrack_proto_tcp.o] Error 1
make[6]: *** [scripts/Makefile.build:503: net/netfilter] Error 2
make[5]: *** [Makefile:1832: net] Error 2
make[5]: Leaving directory '/home/johnny/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.186'
make[4]: *** [Makefile:22: /home/johnny/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.186/.modules] Error 2
make[4]: Leaving directory '/home/johnny/openwrt/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/johnny/openwrt/target/linux'
time: target/linux/compile#1008.00#99.97#1118.79
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/johnny/openwrt'
make[1]: *** [target/Makefile:23: /home/johnny/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/johnny/openwrt'
make: *** [/home/johnny/openwrt/include/toplevel.mk:230：world] 错误 2
22.03.5 内核5.10也出现编译失败 之前没问题

chenmozhijin, 2023-07-18:
[使用方法](https://github.com/chenmozhijin/turboacc/blob/luci/README.md#%E4%BD%BF%E7%94%A8%E6%96%B9%E6%B3%95)已[更新修复](https://github.com/chenmozhijin/turboacc/commit/fba2ae93e4cd75505c464f59d994730a19ccea5b)，并且通过了编译验证[Actions#37](https://github.com/chenmozhijin/turboacc/actions/runs/5585362562)，请尝试使用全新源码按照新的使用方法重新编译。

closed 2023-07-19 by chenmozhijin