chenmozhijin/turboacc #7: [openwrt/23.05]An error occurred while building Netfilter in the linux kernel

lushangkan, 2023-07-17:
**System:** Windows11 WSL Ubuntu22.04
**Openwrt Version:** [Official Openwrt](https://github.com/openwrt/openwrt) / openwrt-23.05
**Building target:** ramips/mt7621/Xiaomi redmi AC2100
**Error log:** [log](https://github.com/openwrt/openwrt)
**Error log summary:**
```
Register multiple callbacks to ct events (NF_CONNTRACK_CHAIN_EVENTS) [N/y/?] (NEW) make[7]: *** [scripts/kconfig/Makefile:77: syncconfig] Error 1
make[6]: *** [Makefile:628: syncconfig] Error 2
make[5]: *** [Makefile:748: include/config/auto.conf.cmd] Error 2

```
**Diffconfig:** [config](https://pastebin.com/DvVNcqMb)

lushangkan, 2023-07-18:
> 如果不需要编译sfe，请删除target/linux/generic/hack-5.15/953-net-patch-linux-kernel-to-support-shortcut-fe.patch

删除后仍然错误

chenmozhijin, 2023-07-18:
我正在测试新的使用方法：5.15内核：
```
            mkdir -p turboacc_tmp ./package/turboacc
            cd turboacc_tmp 
            git clone https://github.com/chenmozhijin/turboacc -b package
            cd ../package/turboacc
            git clone https://github.com/fullcone-nat-nftables/nft-fullcone
            git clone https://github.com/chenmozhijin/turboacc
            mv ./turboacc/luci-app-turboacc ./luci-app-turboacc
            rm -rf ./turboacc
            cd ../..
            cp -f turboacc_tmp/turboacc/hack-5.15/952-add-net-conntrack-events-support-multiple-registrant.patch ./target/linux/generic/hack-5.15/613-netfilter_optional_tcp_window_check.patch
            cp -f turboacc_tmp/turboacc/hack-5.15/953-net-patch-linux-kernel-to-support-shortcut-fe.patch ./target/linux/generic/hack-5.15/953-net-patch-linux-kernel-to-support-shortcut-fe.patch
            cp -f turboacc_tmp/turboacc/pending-5.15/613-netfilter_optional_tcp_window_check.patch ./target/linux/generic/pending-5.15/613-netfilter_optional_tcp_window_check.patch
            rm -rf ./package/libs/libnftnl ./package/network/config/firewall4 ./package/network/utils/nftables
            mkdir -p ./package/network/config/firewall4 ./package/libs/libnftnl ./package/network/utils/nftables
            cp -r ./turboacc_tmp/turboacc/shortcut-fe ./package/turboacc
            cp -RT ./turboacc_tmp/turboacc/firewall4-04a06bd70b9808b14444cae81a2faba4708ee231/firewall4 ./package/network/config/firewall4
            cp -RT ./turboacc_tmp/turboacc/libnftnl-1.2.5/libnftnl ./package/libs/libnftnl
            cp -RT ./turboacc_tmp/turboacc/nftables-1.0.7/nftables ./package/network/utils/nftables
            rm -rf turboacc_tmp
            echo "#  CONFIG_NF_CONNTRACK_CHAIN_EVENTS is not set" >> target/linux/generic/config-5.15
            echo "#  CONFIG_SHORTCUT_FE is not set" >> target/linux/generic/config-5.15
            ./scripts/feeds update -a
            ./scripts/feeds install -a
```
测试进度见 [Action#35](https://github.com/chenmozhijin/turboacc/actions/runs/5585362562)，建议使用全新的源码。

closed 2023-07-19 by chenmozhijin