chenmozhijin/turboacc #8: openwrt 22.03.05 main 分支快照版本掉线问题

cctips, 2023-08-02:
编译后，开启fullnat偶发性掉线，并且必须重启openwrt才可以恢复

chenmozhijin, 2023-08-02:
有系统/内核日志吗？我 23.05.0-rc2 一切正常。

cctips, 2023-08-03:
系统日志没有看，今晚回去编一个，然后做个旁路有挂在那里掉线时我给您提供日志。

cctips, 2023-08-03:
> 有系统/内核日志吗？我 23.05.0-rc2 一切正常。如果装了mwan3好像会不稳定。

目前的情况就是这样子的。全新的 openwrt

[image]

[image]

[openwrt-log.txt](https://github.com/chenmozhijin/turboacc/files/12250853/openwrt-log.txt)

[config.txt](https://github.com/chenmozhijin/turboacc/files/12250866/config.txt)

chenmozhijin, 2023-08-03:
你同时都编译了iptables与nftables等多个防火墙
   ```
   CONFIG_DEFAULT_nftables=y
   CONFIG_PACKAGE_ip6tables-nft=y
   CONFIG_PACKAGE_ip6tables-zz-legacy=y
   CONFIG_PACKAGE_iptables-nft=y
   CONFIG_PACKAGE_iptables-zz-legacy=y
   CONFIG_PACKAGE_xtables-legacy=y
   CONFIG_PACKAGE_xtables-nft=y
   ```
我的> Network> Firewall配置：
[image]
[image]
我的> Kernel modules > Netfilter Extensions配置：
[image]
[image]

顺带提醒一下passwall使用nftables要求dnsmasq版本在2.87以上，openwrt 22.03快照版dnsmasq版本为2.86见[package/network/services/dnsmasq/Makefile](https://github.com/openwrt/openwrt/blob/1ec274a204a65c7f8543cf750a0b027a685818cb/package/network/services/dnsmasq/Makefile#L11C1-L11C27)

也可以看看我的openwrt云编译仓库：[OpenWrt-K](https://github.com/chenmozhijin/OpenWrt-K)

cctips, 2023-08-04:
> 你同时都编译了iptables与nftables等多个防火墙
> 
> ```
> CONFIG_DEFAULT_nftables=y
> CONFIG_PACKAGE_ip6tables-nft=y
> CONFIG_PACKAGE_ip6tables-zz-legacy=y
> CONFIG_PACKAGE_iptables-nft=y
> CONFIG_PACKAGE_iptables-zz-legacy=y
> CONFIG_PACKAGE_xtables-legacy=y
> CONFIG_PACKAGE_xtables-nft=y
> ```
> 
> 我的> Network> Firewall配置： [image] [image] 我的> Kernel modules > Netfilter Extensions配置： [image] [image]
> 
> 也可以看看我的openwrt云编译仓库：[OpenWrt-K](https://github.com/chenmozhijin/OpenWrt-K) OpenWrt-K的完整配置：[config.txt](https://github.com/chenmozhijin/turboacc/files/12252719/config.txt)

我是把能用上的基础组件全部选上了，正常来讲，应该就是 so（动态链接库） 文件虽然我没理解过openwrt文件结构和源代码。

真的有影响吗。 :)

chenmozhijin, 2023-08-04:
你的系统日志
```
Thu Aug  3 12:29:03 2023 kern.err kernel: [   12.084820] xt_DNETMAP: CONFIG_NF_NAT is not available in your kernel, hence this module cannot function.
```
但是：nft-fullcone的Makefile中是添加了CONFIG_NF_NAT=y的，我也不明白为什么会这样
https://github.com/chenmozhijin/turboacc/blob/package/nft-fullcone/Makefile#L23C10-L23C10
我觉得是什么地方冲突了，以前我也遇到了类似的问题，你可以把能不选都不选试试我的，固件没有遇到这样的问题。  

luci-app-turboacc只是一个控制中心，fullcone nat功能是使用[wongsyrone/nft-fullcone](https://github.com/wongsyrone/nft-fullcone)实现的，相关的讨论见Chion82/netfilter-full-cone-nat#42，希望可以帮到你。

cctips, 2023-08-04:
> 你的系统日志
>
> ```
> Thu Aug  3 12:29:03 2023 kern.err kernel: [   12.084820] xt_DNETMAP: CONFIG_NF_NAT is not available in your kernel, hence this module cannot function.
> ``` [...]

好的好的，那我尽可能按照您的方法来做。

你的建议我会采纳，晚上回去按照您的方法重新编译一份，至于passwall 2.86的dnsmasq我装上也没啥影响，干就完了。

cctips, 2023-08-04:
CONFIG_DEFAULT_nftables=y
CONFIG_PACKAGE_ip6tables-nft=y
CONFIG_PACKAGE_ip6tables-zz-legacy=y
CONFIG_PACKAGE_iptables-nft=y
CONFIG_PACKAGE_iptables-zz-legacy=y

这3款防火墙可以共存的对吧。

CONFIG_PACKAGE_xtables 我就不编译到系统中了。

chenmozhijin, 2023-08-04:
我认为，最好不要编译"-zz-legacy"，不然iptables规则与nftables规则可能会同时存在
状态/防火墙 页面会出现以下提示:
```
检测到旧版规则

系统上存在旧版 iptables 规则。 不鼓励混合使用 iptables 和 nftables 规则，这可能会导致流量过滤不完整。
```

我认为这也是你passwall 2.86的dnsmasq你装上也没啥影响的原因，因为它可能使用了iptables。

cctips, 2023-08-04:
> 我认为，最好不要编译"-zz-legacy"，不然iptables规则与nftables规则可能会同时存在 状态/防火墙 页面会出现以下提示:
> 
> ```
> 检测到旧版规则
> 
> 系统上存在旧版 iptables 规则。 不鼓励混合使用 iptables 和 nftables 规则，这可能会导致流量过滤不完整。
> ```
> 
> 我认为这也是你passwall 2.86的dnsmasq你装上也没啥影响的原因，因为它可能使用了iptables。

经过我再一次测试，Shortcut-FE 功能不能开。开了之后其他功能就失效了。

chenmozhijin, 2023-08-04:
不是冲突两者之间的作用是一样的，luci-app-turboacc也是设置防火墙实现的Software flow offloading见：
https://github.com/chenmozhijin/turboacc/blob/bac3ec9a620d7a552e49f8109b40013bd5d215e8/luci-app-turboacc/root/etc/init.d/turboacc#L92C1-L93C63  

此外我自己编译的固件没有这些问题：
[image]
[image]

cctips, 2023-08-04:
> 不是冲突两者之间的作用是一样的，luci-app-turboacc也是设置防火墙实现的Software flow offloading见： https://github.com/chenmozhijin/turboacc/blob/bac3ec9a620d7a552e49f8109b40013bd5d215e8/luci-app-turboacc/root/etc/init.d/turboacc#L92C1-L93C63
> 
> 此外我自己编译的固件没有这些问题： [image] [image]

我现在只是单独编译了 turboacc 其他什么都没有，完全 make menuconfig 默认 只勾选了 turboacc 依然不行。

https://github.com/openwrt/openwrt/tree/openwrt-22.03

这个分支（没有切换commit），您可以试试是否和我存在同样的情况。

chenmozhijin, 2023-08-04:
我试着云编译一下：https://github.com/chenmozhijin/turboacc/actions/runs/5765295629
云编译编译了四种固件 OpenWrt-dnsmasq-full-old应该是最有希望的你可以试一下

我现在也搞不清是什么问题了
Software flow offloading调用的是openwrt官方firewall4自带依赖kmod-nft-offload
https://github.com/openwrt/openwrt/blob/1eb67cb0704a61f49d9be9f88aecbce1d881f8c9/package/network/config/firewall4/Makefile#L25
应该不会出问题。
有可能是修补的firewall4、libnftnl、nftables太新了
使用方法中的版本目前分别为
FIREWALL4_VERSION=04a06bd70b9808b14444cae81a2faba4708ee231
NFTABLES_VERSION=1.0.8
LIBNFTNL_VERSION=1.2.6
而openwrt-22.03分支为
FIREWALL4_VERSION=7ae5e14bbd7265cc67ec870c3bb0c8e197bb7ca9
NFTABLES_VERSION=1.0.2
LIBNFTNL_VERSION=1.2.1
本仓库package分支包含修补旧版的firewall4、libnftnl、nftables你可以试着替换一下

chenmozhijin, 2023-08-04:
我试着云编译一下：https://github.com/chenmozhijin/turboacc/actions/runs/5765352624
云编译编译了四种固件 OpenWrt-dnsmasq-full-old应该是最有希望的你可以试一下

我现在也搞不清是什么问题了
Software flow offloading调用的是openwrt官方firewall4自带依赖kmod-nft-offload
https://github.com/openwrt/openwrt/blob/1eb67cb0704a61f49d9be9f88aecbce1d881f8c9/package/network/config/firewall4/Makefile#L25
应该不会出问题。
有可能是修补的firewall4、libnftnl、nftables太新了
使用方法中的版本目前分别为
FIREWALL4_VERSION=04a06bd70b9808b14444cae81a2faba4708ee231
NFTABLES_VERSION=1.0.8
LIBNFTNL_VERSION=1.2.6
而openwrt-22.03分支为
FIREWALL4_VERSION=7ae5e14bbd7265cc67ec870c3bb0c8e197bb7ca9
NFTABLES_VERSION=1.0.2
LIBNFTNL_VERSION=1.2.1
本仓库package分支包含修补旧版的firewall4、libnftnl、nftables你可以试着替换一下

cctips, 2023-08-04:
> 我试着云编译一下：https://github.com/chenmozhijin/turboacc/actions/runs/5765352624 云编译编译了四种固件 OpenWrt-dnsmasq-full-old应该是最有希望的你可以试一下
>
> 我现在也搞不清是什么问题了 Software flow [...]

```
mkdir -p turboacc_tmp ./package/turboacc
cd turboacc_tmp 
git clone https://github.com/chenmozhijin/turboacc -b package
cd ../package/turboacc
git clone https://github.com/fullcone-nat-nftables/nft-fullcone
git clone https://github.com/chenmozhijin/turboacc
mv ./turboacc/luci-app-turboacc ./luci-app-turboacc
rm -rf ./turboacc
cd ../..
cp -f turboacc_tmp/turboacc/hack-5.10/952-net-conntrack-events-support-multiple-registrant.patch ./target/linux/generic/hack-5.10/952-net-conntrack-events-support-multiple-registrant.patch
cp -f turboacc_tmp/turboacc/hack-5.10/953-net-patch-linux-kernel-to-support-shortcut-fe.patch ./target/linux/generic/hack-5.10/953-net-patch-linux-kernel-to-support-shortcut-fe.patch
cp -f turboacc_tmp/turboacc/pending-5.10/613-netfilter_optional_tcp_window_check.patch ./target/linux/generic/hack-5.10/613-netfilter_optional_tcp_window_check.patch
rm -rf ./package/libs/libnftnl ./package/network/config/firewall4 ./package/network/utils/nftables
mkdir -p ./package/network/config/firewall4 ./package/libs/libnftnl ./package/network/utils/nftables
cp -r ./turboacc_tmp/turboacc/shortcut-fe ./package/turboacc
cp -RT ./turboacc_tmp/turboacc/firewall4-7ae5e14bbd7265cc67ec870c3bb0c8e197bb7ca9/firewall4 ./package/network/config/firewall4
cp -RT ./turboacc_tmp/turboacc/libnftnl-1.2.1/libnftnl ./package/libs/libnftnl
cp -RT ./turboacc_tmp/turboacc/nftables-1.0.2/nftables ./package/network/utils/nftables
rm -rf turboacc_tmp
echo "# CONFIG_NF_CONNTRACK_CHAIN_EVENTS is not set" >> target/linux/generic/config-5.10
echo "# CONFIG_SHORTCUT_FE is not set" >> target/linux/generic/config-5.10
./scripts/feeds update -a
./scripts/feeds install -a
```

经过这一夜的测试，还有你的建议，均无效。我好累，去睡了。

cctips, 2023-08-05:
> 我试着云编译一下：https://github.com/chenmozhijin/turboacc/actions/runs/5765352624 云编译编译了四种固件 OpenWrt-dnsmasq-full-old应该是最有希望的你可以试一下
>
> 我现在也搞不清是什么问题了 Software flow [...]

两个视频，一个文件太大压缩了一下。测试云编译的也均无效。

https://github.com/chenmozhijin/turboacc/assets/45416520/481c0d4d-65ee-4aef-9053-793d78f7a1b9

[openwrt-dnsmasq_2.zip](https://github.com/chenmozhijin/turboacc/files/12265399/openwrt-dnsmasq_2.zip)

chenmozhijin, 2023-08-05:
这应该是因为没有依赖luci、luci-compat的所以没有编译lua语言支持的所导致的，
我已经加上了 76af113f17b699c2123a482a1aaba2b228612b30
现在在重新编译：https://github.com/chenmozhijin/turboacc/actions/runs/5768775903

cctips, 2023-08-05:
> 这应该是因为没有依赖luci、luci-compat的所以没有编译lua语言支持的所导致的， 我已经加上了 [76af113](https://github.com/chenmozhijin/turboacc/commit/76af113f17b699c2123a482a1aaba2b228612b30) 现在在重新编译：https://github.com/chenmozhijin/turboacc/actions/runs/5768775903 我觉得应该和dnsmasq的关系不大， 重新编译完之后请试试：OpenWrt-dnsmasq-old

你说的这个依赖我安装了。 **opkg install luci luci-base luci-compat**

chenmozhijin, 2023-08-05:
开
uci set firewall.@defaults[0].flow_offloading="1"
uci set firewall.@defaults[0].flow_offloading_hw="1"
uci set firewall.@defaults[0].fullcone="1"
uci commit firewall
sysctl -w net.ipv4.tcp_congestion_control="bbr"
/etc/init.d/dnsmasq restart
/etc/init.d/firewall restart
关
uci set firewall.@defaults[0].flow_offloading="0"
uci set firewall.@defaults[0].flow_offloading_hw="0"
uci set firewall.@defaults[0].fullcone="0"
uci commit firewall
sysctl -w net.ipv4.tcp_congestion_control="cubic"
/etc/init.d/dnsmasq restart
/etc/init.d/firewall restart
直接终端执行看看有没有效果

cctips, 2023-08-05:
> 开 uci set firewall.@defaults[0].flow_offloading="1" uci set firewall.@defaults[0].flow_offloading_hw="1" uci set firewall.@defaults[0].fullcone="1" uci commit firewall sysctl -w net.ipv4.tcp_congestion_control="bbr" /etc/init.d/dnsmasq restart /etc/init.d/firewall restart 关 uci set firewall.@defaults[0].flow_offloading="0" uci set firewall.@defaults[0].flow_offloading_hw="0" uci set firewall.@defaults[0].fullcone="0" uci commit firewall sysctl -w net.ipv4.tcp_congestion_control="cubic" /etc/init.d/dnsmasq restart /etc/init.d/firewall restart 直接终端执行看看有没有效果，我这里测试的是可以正常开关功能的
> 
> 使用turboacc 脚本测试 开 uci set turboacc.config.sw_flow='1' uci set turboacc.config.hw_flow='1' uci set turboacc.config.fullcone_nat='1' uci set turboacc.config.bbr_cca='1' uci commit turboacc /etc/init.d/turboacc start 关 uci set turboacc.config.sw_flow='0' uci set turboacc.config.hw_flow='0' uci set turboacc.config.fullcone_nat='0' uci set turboacc.config.bbr_cca='0' uci commit turboacc /etc/init.d/turboacc start

命令开关是好用的，确实好使。

cctips, 2023-08-07:
dnsmasq 如何替换成 更高的版本呢。我记得你发过一个版本号的 makefile 文件连接，找不到。

好像是防火墙的问题造成第一项没法开启。 dnsmasq 2.87 以上会有更好的支持。

chenmozhijin, 2023-08-07:
dnsmasq
https://github.com/openwrt/openwrt/tree/master/package/network/services/dnsmasq
libubox
https://github.com/openwrt/openwrt/tree/main/package/libs/libubox

要升级的话上面两个都要替换，不过直接用openwrt-23.05分支不好吗？

cctips, 2023-08-07:
> dnsmasq https://github.com/openwrt/openwrt/tree/master/package/network/services/dnsmasq libubox https://github.com/openwrt/openwrt/tree/main/package/libs/libubox
> 
> 要升级的话上面两个都要替换，不过直接用openwrt-23.05分支不好吗？
> 
> OpenWrt-K自带这项功能可以参考一下： https://github.com/chenmozhijin/OpenWrt-K/blob/0c819e0b8b033c203baf42d5cace5b217c03832b/.github/workflows/build-openwrt.yml#L210C1-L211C96 https://github.com/chenmozhijin/OpenWrt-K/blob/0c819e0b8b033c203baf42d5cace5b217c03832b/.github/workflows/build-openwrt.yml#L465C1-L477C13

好的，那就直接升级吧，顺便把您给我的资料记录一下。方便以后参考用，。

fuqiang03, 2023-08-09:
建议不要升级,建议停用一些开机自启动的脚本 或者 rmod 一些 不用的模块
[image]
越折腾越不稳定,越不稳定越折腾

chenmozhijin, 2023-08-10:
我用自己云编译的v23.05.0-rc2版本并没有遇到什么问题，已经13天没管过了。

升级是因为22.03.05自带的版本为dnsmasq版本为2.86，nftables透明代理要求dnsmasq版本在2.87以上并开启nftset支持，不升级无法正常使用passwall。

[image]

fuqiang03, 2023-08-10:
用的pssswall 2 ?    我现在用dsnmaaq2.86 +pssswall 貌似没有问题  因为使用的4.14内核 对fw4+nft 还在观望
{运行的效果不理想,部分插件也不支持(使用5.15内核,实测同样的插件游戏的延迟会增加} 相对fw3 +iptables 已经很好了
[image]

以下的屎山,能否好些

unload_sfe() {
	flow_enabled=$(uci get flowoffload.@flow[0].flow_offloading 2>/dev/null)

	if [ $flow_enabled -eq 0 -o $flow_enabled -gt 1 ]; then
		uci set firewall.@defaults[0].flow_offloading=$(uci get flowoffload.@flow[0].flow_offloading)
		uci set firewall.@defaults[0].flow_offloading_hw=$(uci get flowoffload.@flow[0].flow_offloading_hw)
		uci commit firewall
		logger -t alex  $flow_enabled stop-down flowoffload
	fi

	if [ -d /sys/module/shortcut_fe ]; then
		logger -t alex  file exists shortcut-fe and remove Shortcut-fe.
		rmmod shortcut-fe
	fi

	if [ -d /sys/module/shortcut_fe_ipv6 ]; then
		logger -t alex  file exists shortcut-fe-ipv6 and remove shortcut-fe-ipv6.
		rmmod shortcut_fe_ipv6
	fi

	if [ -d /sys/module/fast_classifier ]; then
		logger -t alex  file exists fast_classifier and remove fast_classifier.
		rmmod fast_classifier
	fi
}

start(){
	local kernel_version=$(uname -r)

	bbr_enable=$(uci get flowoffload.@flow[0].bbr_enabled 2>/dev/null)
	nat1_enable=$(uci get flowoffload.@flow[0].nat1_enabled 2>/dev/null)
	dns=$(uci get flowoffload.@flow[0].dns 2>/dev/null)
	flow_enabled=$(uci get flowoffload.@flow[0].flow_offloading 2>/dev/null)

	if [ $bbr_enable -eq 1 ];  then
		start_bbr
	fi
	if [ $nat1_enable -gt 0 ];  then
		start_fullconenat
	fi
	echo  "$flow_enabled"

	if [ $flow_enabled -eq 0 ]; then
	unload_sfe
	fi
	if [ $flow_enabled -eq 1 ]; then
	uci set firewall.@defaults[0].flow_offloading=$(uci get flowoffload.@flow[0].flow_offloading)
	uci set firewall.@defaults[0].flow_offloading_hw=$(uci get flowoffload.@flow[0].flow_offloading_hw)
	uci commit firewall
	elif [ $flow_enabled -eq 2 ]; then
	[ -e "/lib/modules/$kernel_version/shortcut-fe.ko" ] && modprobe shortcut-fe
	logger -t alex  $flow_enabled start-up shortcut-fe
	elif [ $flow_enabled -eq 3  ];then
	[ -e "/lib/modules/$kernel_version/fast-classifier.ko" ] && modprobe fast-classifier
	logger -t alex  $flow_enabled start-up fast-classifier
	fi

	logger -t alex  "flowoffload command starting success"
}

chenmozhijin, 2023-08-10:
> 用的pssswall 2 ？我现在用dsnmaaq2.86 +pssswall 貌似没有问题 因为使用的4.14内核 对fw4+nft 还在观望 {运行的效果不理想，部分插件也不支持（使用5.15内核，实测同样的插件游戏的延迟会增加} 相对fw3 +iptables 已经很好了 [image]
> 
> 以下的屎山,能否好些
> 
> unload_sfe（） { flow_enabled=$（uci get flowoffload.[0].flow_offloading 2>/dev/null）
> 
> ```
> if [ $flow_enabled -eq 0 -o $flow_enabled -gt 1 ]; then
> 	uci set firewall.@defaults[0].flow_offloading=$(uci get flowoffload.@flow[0].flow_offloading)
> 	uci set firewall.@defaults[0].flow_offloading_hw=$(uci get flowoffload.@flow[0].flow_offloading_hw)
> 	uci commit firewall
> 	logger -t alex  $flow_enabled stop-down flowoffload
> fi
> 
> if [ -d /sys/module/shortcut_fe ]; then
> 	logger -t alex  file exists shortcut-fe and remove Shortcut-fe.
> 	rmmod shortcut-fe
> fi
> 
> if [ -d /sys/module/shortcut_fe_ipv6 ]; then
> 	logger -t alex  file exists shortcut-fe-ipv6 and remove shortcut-fe-ipv6.
> 	rmmod shortcut_fe_ipv6
> fi
> 
> if [ -d /sys/module/fast_classifier ]; then
> 	logger -t alex  file exists fast_classifier and remove fast_classifier.
> 	rmmod fast_classifier
> fi
> ```
> 
> }
> 
> start（）{ local kernel_version=$（uname -r）
> 
> ```
> bbr_enable=$(uci get flowoffload.@flow[0].bbr_enabled 2>/dev/null)
> nat1_enable=$(uci get flowoffload.@flow[0].nat1_enabled 2>/dev/null)
> dns=$(uci get flowoffload.@flow[0].dns 2>/dev/null)
> flow_enabled=$(uci get flowoffload.@flow[0].flow_offloading 2>/dev/null)
> 
> if [ $bbr_enable -eq 1 ];  then
> 	start_bbr
> fi
> if [ $nat1_enable -gt 0 ];  then
> 	start_fullconenat
> fi
> echo  "$flow_enabled"
> 
> if [ $flow_enabled -eq 0 ]; then
> unload_sfe
> fi
> if [ $flow_enabled -eq 1 ]; then
> uci set firewall.@defaults[0].flow_offloading=$(uci get flowoffload.@flow[0].flow_offloading)
> uci set firewall.@defaults[0].flow_offloading_hw=$(uci get flowoffload.@flow[0].flow_offloading_hw)
> uci commit firewall
> elif [ $flow_enabled -eq 2 ]; then
> [ -e "/lib/modules/$kernel_version/shortcut-fe.ko" ] && modprobe shortcut-fe
> logger -t alex  $flow_enabled start-up shortcut-fe
> elif [ $flow_enabled -eq 3  ];then
> [ -e "/lib/modules/$kernel_version/fast-classifier.ko" ] && modprobe fast-classifier
> logger -t alex  $flow_enabled start-up fast-classifier
> fi
> 
> logger -t alex  "flowoffload command starting success"
> ```
> 
> }

pssswall是防火墙工具使用nftables时需要nftables透明代理，fw3 + iptables的话pssswall用的是iptables与ipset。  
本插件使用[nft-fullcone](https://github.com/fullcone-nat-nftables/nft-fullcone)实现全锥型 NAT，应该不支持fw3？  

您这是把[luci-app-flowoffload](https://github.com/coolsnowwolf/openwrt/tree/lede-17.01/package/lean/luci-app-flowoffload)里面的代码结合进来了？  

https://github.com/chenmozhijin/turboacc/blob/bee265a11e0bba6af2a8252cc48acf62c3263b6d/luci-app-turboacc/root/etc/init.d/turboacc?plain=1#L14C1-L20C55

 这一段就已经获取配置到变量了，您可以参考一下openwrt的文档：
[The UCI system](https://openwrt.org/docs/guide-user/base-system/uci)与[Configuration in scripts](https://openwrt.org/docs/guide-developer/config-scripting)
您这样获取的是luci-app-flowoffload的配置而不是本插件的

[cctips](https://github.com/cctips)遇到的应该是两个问题，一开始应该是防火墙的兼容性问题，后面是luci-app-turboacc无法正常开关功能的问题。插件在我的固件上是一切正常，我无法成功复现出问题。

chenmozhijin, 2023-08-11:
我只知道shortcut-fe是fast-classifier的依赖，我不用sfe用firewall4自带的flow offload所以不太清楚。不过lede原本的[luci-app-turboacc](https://github.com/coolsnowwolf/luci/tree/master/applications/luci-app-turboacc)就是这样写的

您uci get得到的是flowoffload的配置，可以参考https://openwrt.org/docs/guide-user/base-system/uci#usage。

fuqiang03, 2023-08-11:
好的 我会参照uci标准 修改config获取变量的方式的 在多问一下 有4.14内核 最新的shortcut补丁吗 有的话可否分享下

chenmozhijin, 2023-08-11:
> 好的 我会参照uci标准 修改config获取变量的方式的 在多问一下 有4.14内核 最新的shortcut补丁吗 有的话可否分享下

补丁：
https://github.com/coolsnowwolf/lede/blob/20200915/target/linux/generic/hack-4.14/952-net-conntrack-events-support-multiple-registrant.patch
https://github.com/coolsnowwolf/lede/blob/20200915/target/linux/generic/hack-4.14/999-net-patch-linux-kernel-to-support-shortcut-fe.patch

closed 2023-09-26 by chenmozhijin