chenmozhijin/turboacc #10: Turbo acc not working with latest OpenWrt Snapshot.

butinhi, 2023-08-23:
I turn on all but not working

[url=https://ibb.co/232Xq9h][img]https://i.ibb.co/232Xq9h/Screenshot-from-2023-08-23-13-37-09.png[/img][/url] [url=https://ibb.co/TgS3wVT][img]https://i.ibb.co/TgS3wVT/Screenshot-from-2023-08-23-13-37-30.png[/img][/url]

chenmozhijin, 2023-08-23:
Please provide your system logs and device architecture.

Please try to enable only Software flow offloading, FullCone NAT and BBR CCA.

You can try to execute the following command in terminal to see if it works
```sh
uci set firewall.@defaults[0].flow_offloading="1"
uci set firewall.@defaults[0].flow_offloading_hw="1"
uci set firewall.@defaults[0].fullcone="1"
uci commit firewall
sysctl -w net.ipv4.tcp_congestion_control="bbr"
/etc/init.d/dnsmasq restart
/etc/init.d/firewall restart
```
If possible, please provide the output after execution.

chenmozhijin, 2023-08-23:
I have not found the problem in the syslog so far.

You can try to execute the following command in terminal to see if it works

uci set firewall.@defaults[0].flow_offloading="1"
uci set firewall.@defaults[0].flow_offloading_hw="1"
uci set firewall.@defaults[0].fullcone="1"
uci commit firewall
sysctl -w net.ipv4.tcp_congestion_control="bbr"
/etc/init.d/dnsmasq restart
/etc/init.d/firewall restart

please provide the output after execution.

butinhi, 2023-08-23:
The same 
[image]

chenmozhijin, 2023-08-23:
please provide the output after execution.

butinhi, 2023-08-23:
Not like you 

[image]

chenmozhijin, 2023-08-23:
Full cone NAT requires patching firewall4, libnftnl, nftables.
I think you may not have patched successfully.

butinhi, 2023-08-23:
> Full cone NAT requires patching firewall4, libnftnl, nftables. I think you may not have patched successfully.

I patched like your guide

butinhi, 2023-08-23:
Do you use telegram? Pls help me via telegram: https://t.me/dvh151

chenmozhijin, 2023-08-23:
I just updated the openwrt version to v23.05.0-rc3, currently there are only 5 new commits compared to 23.05-snapshot, and I tested without any problems.
[image]
[image]
[image]

I think you may not have successfully patched firewall4, libnftnl, nftables.
Please check whether the following files exist in the openwrt compilation root directory
1. package/network/config/firewall4/patches
/999-01-firewall4-add-fullcone-support.patch
2. package/libs/libnftnl/patches
/999-01-libnftnl-add-fullcone-expression-support.patch
3. package/network/utils/nftables/patches
/999-01-nftables-add-fullcone-expression-support.patch

butinhi, 2023-08-23:
My AX6s working with lede openwrt. Its turboacc has this option 
[image]

butinhi, 2023-08-23:
Maybe I will change to lede. Its good for my device :(

chenmozhijin, 2023-08-23:
Are you sure you patched firewall4, libnftnl, nftables successfully?
If the patch was successful, there should be output when restarting the firewall.
Please check whether the following files exist in the openwrt compilation root directory
1. package/network/config/firewall4/patches
/999-01-firewall4-add-fullcone-support.patch
2. package/libs/libnftnl/patches
/999-01-libnftnl-add-fullcone-expression-support.patch
3. package/network/utils/nftables/patches
/999-01-nftables-add-fullcone-expression-support.patch

It is really convenient to compile turboacc in lede, but the official source code also has its advantages, just choose the one that suits you.

butinhi, 2023-08-23:
> Are you sure you patched firewall4, libnftnl, nftables successfully? If the patch was successful, there should be output when restarting the firewall. Please check whether the following files exist [...]

D0 you use telegram?

butinhi, 2023-08-24:
> Are you sure you patched firewall4, libnftnl, nftables successfully? If the patch was successful, there should be output when restarting the firewall. Please check whether the following files exist [...]

[image] [image] [image]

chenmozhijin, 2023-08-24:
Can you provide the .config file for compiling openwrt?

butinhi, 2023-08-24:
> Can you provide the .config file for compiling openwrt?

I can not see it on Ubuntu

chenmozhijin, 2023-08-25:
Please try to turn off software flow offloading.

butinhi, 2023-08-25:
> Please try to turn off software flow offloading.

Tried not but not working

chenmozhijin, 2023-08-25:
Openwrt official documentation:
https://openwrt.org/toh/xiaomi/ax3200#verifying_that_wed_is_enabled

If verification shows that WEB is enabled, there may be other issues.

lede uses firewall3 while openwrt 23.05 uses firewall4, which may affect performance.