chenmozhijin/turboacc #14: FullCone NAT 无法开启

LostAttractor, 2023-11-03:
[image]
OpenWRT主线，前两天最新的master分支，没打其他的任何patch，几周前第一次尝试这个插件就有此问题，一直是master分支
[image]
似乎nft_fullcone这个内核模块是挂上了的
[image]
稍微读了下源码，然后发现`/sys/module/nft_fullcone/refcnt`是0，不知道是不是应该把issues发到nft-fullcone这个项目那边去
`dmesg | grep fullcone`和`logread | grep fullcone`的结果都是空的，我不知道我应该去哪里找日志，如果有日志的话，还请告诉我

以下是我编译时安装的一些插件
```
luci

luci-app-acme
luci-app-ddns
luci-app-nlbwmon
luci-app-passwall2
luci-app-smartdns
luci-app-sqm
luci-app-statistics
luci-app-upnp
luci-app-wol

luci-theme-argon

luci-proto-wireguard

bind-dig
bind-host
bind-nslookup

ncat-full
nmap-full

acme-acmesh-dnsapi
ethtool-full
iperf3
tcpdump

nano-full
vim-full

acl
btrfs-progs
- Built with zstd support
ntfs-3g

dmesg
hwinfo
irqbalance
qrencode
smartmontools
whereis
which

collectd
- Enable ability to use with encrypted networking
collectd-mod-dns
collectd-mod-ethstat
collectd-mod-interface
collectd-mod-ping
collectd-mod-sensors
collectd-mod-thermal
collectd-mod-wireless

python3-speedtest-cli⏎                         
```

chenmozhijin, 2023-11-03:
`/etc/init.d/firewall restart`重启一下防火墙看看有什么输出

[image]

LostAttractor, 2023-11-04:
谢谢你提供思路！
[image]
曾经装的openclash，passwall1留下了很多残留的include
```
config include 'openclash'
	option type 'script'
	option path '/var/etc/openclash.include'

config include 'passwall'
	option type 'script'
	option path '/var/etc/passwall.include'
	option reload '1'

config include 'passwall_server'
	option type 'script'
	option path '/var/etc/passwall_server.include'
	option reload '1'
	
config include 'passwall2'
	option type 'script'
	option path '/var/etc/passwall2.include'
	option reload '1'

config include 'passwall2_server'
	option type 'script'
	option path '/var/etc/passwall2_server.include'
	option reload '1'
```
恢复成默认后问题就得到了解决
[image]
TurboACC插件内显示的也正常了
不过passwall和openclash的残留还能够理解
为什么passwall2会加入一个不存在的`passwall2_server`的include，我并没有开启passwall的server功能，include的这个文件也因此是不存在的

chenmozhijin, 2023-11-04:
不清楚，应该是passwall2在[UCI defaults](https://openwrt.org/docs/guide-developer/uci-defaults)添加了
https://github.com/xiaorouji/openwrt-passwall2/blob/d73fb54249a3c9862f34a1124c05069e62ef22ce/luci-app-passwall2/root/etc/uci-defaults/luci-passwall2#L20-L25

不过我看你开启了IPv6 全锥形 ，NAT这会给IPv6增加一层NAT https://github.com/chenmozhijin/OpenWrt-K/issues/39 一般来说应该是不需要开启的。

LostAttractor, 2023-11-04:
> 不清楚，应该是passwall2在[UCI defaults](https://openwrt.org/docs/guide-developer/uci-defaults)添加了 https://github.com/xiaorouji/openwrt-passwall2/blob/d73fb54249a3c9862f34a1124c05069e62ef22ce/luci-app-passwall2/root/etc/uci-defaults/luci-passwall2#L20-L25
> 
> 不过我看你开启了IPv6 全锥形 ，NAT这会给IPv6增加一层NAT [chenmozhijin/OpenWrt-K#39](https://github.com/chenmozhijin/OpenWrt-K/issues/39) 一般来说应该是不需要开启的。

我当然不会给v6开NAT，但是openwrt编译出的image里的`/etc/config/firewall`中，fullcone6默认就是开启的，像这样
```
config zone
	option name		wan
	list   network		'wan'
	list   network		'wan6'
	option input		REJECT
	option output		ACCEPT
	option forward		REJECT
	option fullcone4 '1'
	option fullcone6 '1'
	option masq		1
	option mtu_fix		1
```
重启turboacc就恢复了

closed 2023-11-04 by LostAttractor