chenmozhijin/turboacc #15: Error on compiling

SuperKali, 2023-12-17:
Hello,

I experiencing a error when I compile this with openwrt 23.05.0/23.05.2:

[image]

I followed step by step your guide but I can't compile the code.

Can you help me?

chenmozhijin, 2023-12-20:
I tested on openwrt 23.05.2 and the problem you encountered did not occur
https://github.com/chenmozhijin/turboacc/actions/runs/7265041893/job/19793888396

You can try cloning the openwrt repository again and recompiling to see if this problem still exists

SuperKali, 2023-12-20:
Hello @chenmozhijin, this is very strange i tried a couple time by cloning OpenWrt and i get anyway the error.

Can you confirm that the guide is correct on readme?

SuperKali, 2023-12-21:
[image]

Still getting error, i have tried with a fresh compile i'm using debian 11

SuperKali, 2023-12-21:
[image]

chenmozhijin, 2023-12-21:
I think this error may be related to your openwrt configuration.

I updated the guide so that now you only need to run a script to add turboacc to openwrt. And the workflow for testing will use the method in the guide to add turboacc.

https://github.com/chenmozhijin/turboacc/blob/5a5f03bfc8f5b439abdda8cc279c052be0d7ef73/.github/workflows/test.yml#L111

Now that the test is running, if the error you encountered does not appear, the error is most likely related to your configuration.
https://github.com/chenmozhijin/turboacc/actions/runs/7290986833

Compiling fast-classifier is not necessary for luci-app-turboacc. You can cancel compilation by deselecting "Include Shortcut-FE"

SuperKali, 2023-12-21:
i will try now thanks

closed 2023-12-27 by chenmozhijin