chenmozhijin/turboacc #18: 找不到fullcone的选项

wumode, 2024-01-12:
使用immortalwrt的openwrt-23.05源码按照README的使用方法编译，没有fullcone的选项
[image]
执行`/etc/init.d/firewall restart`显示：
```
root@N1:~# /etc/init.d/firewall restart
nft_try_fullcone: cmd /usr/sbin/nft -c 'add table inet fw4-fullcone-test; add chain inet fw4-fullcone-test dstnat { type nat hook prerouting priority -100; policy accept; fullcone; }; add chain inet fw4-fullcone-test srcnat { type nat hook postrouting priority -100; policy accept; fullcone; }; ' 2>/dev/null
nft_try_fullcone failed, disable fullcone globally
nft_try_fullcone: cmd /usr/sbin/nft -c 'add table inet fw4-fullcone-test; add chain inet fw4-fullcone-test dstnat { type nat hook prerouting priority -100; policy accept; fullcone; }; add chain inet fw4-fullcone-test srcnat { type nat hook postrouting priority -100; policy accept; fullcone; }; ' 2>/dev/null
nft_try_fullcone failed, disable fullcone globally
Section @zone[0] (lan) fullcone in defaults not enabled, ignore zone fullcone settings
Section @zone[1] (wan) fullcone in defaults not enabled, ignore zone fullcone settings
Section docker (docker) fullcone in defaults not enabled, ignore zone fullcone settings
Section @rule[9] (Support-UDP-Traceroute) is disabled, ignoring section
Section @include[0] is not marked as compatible with fw4, ignoring section
Section @include[0] requires 'option fw4_compatible 1' to be considered compatible
Section miniupnpd option 'family' is not supported by fw4
Section miniupnpd option 'reload' is not supported by fw4
Section miniupnpd specifies unreachable path '/usr/share/miniupnpd/firewall.include', ignoring section
Section koolproxy option 'reload' is not supported by fw4
Section koolproxy specifies unreachable path '/var/etc/koolproxy.include', ignoring section
Section pptpd option 'reload' is not supported by fw4
Section pptpd specifies unreachable path '/etc/pptpd.include', ignoring section
Section vssr option 'reload' is not supported by fw4
Section vssr specifies unreachable path '/var/etc/vssr.include', ignoring section
Automatically including '/usr/share/nftables.d/table-post/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/dstnat/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/forward/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/srcnat/20-miniupnpd.nft'

```
请问一下这是什么情况?

chenmozhijin, 2024-01-12:
是不是没有安装kmod-nft-fullcone？

luci-app-turboacc会检查"/lib/modules/你的内核版本/nft_fullcone.ko"文件否存在（即nft-fullcone内核模块），不存在不会显示fullcone的选项

wumode, 2024-01-12:
确实没有nft_fullcone.ko这个文件，而且在执行make menuconfig时会显示错误
```
tmp/.config-package.in:65220:error: recursive dependency detected!
tmp/.config-package.in:65220:   symbol PACKAGE_luci-app-turboacc depends on PACKAGE_kmod-nft-fullcone
tmp/.config-package.in:15528:   symbol PACKAGE_kmod-nft-fullcone is selected by PACKAGE_luci-app-turboacc
```
kmod-nft-fullcone被选中了，但是有两个kmod-nft-fullcone会被同时取消和选中
[image]
请问这是为什么啊？

chenmozhijin, 2024-01-12:
immortalwrt 好像自带了一个nft-fullcone https://github.com/immortalwrt/immortalwrt/blob/openwrt-23.05/package/network/utils/fullconenat-nft 把它删了看看

wumode, 2024-01-13:
感谢！把immortalwrt自带的nft-fullcone删了，再重编译就没问题了

closed 2024-01-13 by wumode