chenmozhijin/turboacc #19: 编译成功了，也刷进去了，但turboacc一直显示“正在收集数据”。。。红米AX6s，官方op22.03分支

hpx502766238, 2024-02-01:
编译成功了，也刷进去了，但点开turboacc一直显示“正在收集数据”。。。红米AX6s，官方op22.03分支

chenmozhijin, 2024-02-02:
截张图看看
重启一下试试
运行
```
/etc/init.d/firewall restart  
cat "/sys/module/nft_flow_offload/refcnt"
cat "/proc/sys/net/ipv4/tcp_congestion_control"
cat "/sys/module/nft_fullcone/refcnt"
```
看看输出

hpx502766238, 2024-02-02:
> 截张图看看 重启一下试试 运行
> 
> ```
> /etc/init.d/firewall restart  
> cat "/sys/module/nft_flow_offload/refcnt"
> cat "/proc/sys/net/ipv4/tcp_congestion_control"
> cat "/sys/module/nft_fullcone/refcnt"
> ```
> 
> 看看输出

我昨天折腾了一晚上，最后一遍成功了，最后一遍时候，我先克隆 了 op22.03，然后打这个补丁，然后删掉 953 的 sfe  patch，第二步覆盖闭源无线驱动，第三步更新和安装 feed，makemenuconfig， 重新编译之后不保留配置刷入就好了。

hpx502766238, 2024-02-02:
> 我昨天折腾了一晚上，最后一遍成功了，最后一遍时候，我先克隆 了 op22.03，然后打这个补丁，然后删掉 953 的 sfe patch，第二步覆盖闭源无线驱动，第三步更新和安装 feed，makemenuconfig， 重新编译之后不保留配置刷入就好了。

我不确定的就是这个会不会和闭源驱动有冲突，，虽然没有覆盖相同的文件

chenmozhijin, 2024-02-02:
这我也不太清楚，现在我是用的x86-64架构的23.05什么问题都没有，以前22.03的时候好像也遇到过这种情况，我也忘了我是怎么搞好的了。

hpx502766238, 2024-02-03:
> 这我也不太清楚，现在我是用的x86-64架构的23.05什么问题都没有，以前22.03的时候好像也遇到过这种情况，我也忘了我是怎么搞好的了。

我找到原因了。当固件里面使用firewall4、nftable同时又存在iptable时候，固件会优先运行nftable，而插件优先识别根本不运行的iptable导致识别失败。至于为什么编译时选了iptable，是因为选dnsmasq-full的时候勾选了iptable支持，所以强行装上了iptable

chenmozhijin, 2024-02-03:
> > 这我也不太清楚，现在我是用的x86-64架构的23.05什么问题都没有，以前22.03的时候好像也遇到过这种情况，我也忘了我是怎么搞好的了。
> 
> 我找到原因了。当固件里面使用firewall4、nftable同时又存在iptable时候，固件会优先运行nftable，而插件优先识别根本不运行的iptable导致识别失败。至于为什么编译时选了iptable，是因为选dnsmasq-full的时候勾选了iptable支持，所以强行装上了iptable

可是这插件里有没有一行有关iptable的操作啊，对防火墙的操作都是用的openwrt自带防火墙（firewall4）

[image]

下面是检查运行状态的代码
https://github.com/chenmozhijin/turboacc/blob/98e280a5b835352b454d21567aa24b990c30f859/luci-app-turboacc/root/etc/init.d/turboacc#L129-L170
这跟firewall4、nftable、iptable应该都没有什么关系，所以这个问题还是很奇怪

hpx502766238, 2024-02-07:
> 可是这插件里有没有一行有关iptable的操作啊，对防火墙的操作都是用的openwrt自带防火墙（firewall4） [...]

又测试了一波，好像我判断错了，即使被强行装上kmod-ipt-core，现在Turboacc也不出问题了，但现在有很多其他问题，很多插件开启无效，比如UU加速器无法启动，UPNP开启之后映射列表一条都没有貌似也没有生效，weburl也是开启没有任何过滤效果，nft-qos也没有限速效果，不知道是不是跟防火墙有关系，这要怎么解决呢

chenmozhijin, 2024-02-07:
把 全锥型 NAT 关了看看

hpx502766238, 2024-02-07:
> 把 全锥型 NAT 关了看看

没啥用，还是老样子

mis1042, 2024-02-11:
> > 这我也不太清楚，现在我是用的x86-64架构的23.05什么问题都没有，以前22.03的时候好像也遇到过这种情况，我也忘了我是怎么搞好的了。
> 
> 我找到原因了。当固件里面使用firewall4、nftable同时又存在iptable时候，固件会优先运行nftable，而插件优先识别根本不运行的iptable导致识别失败。至于为什么编译时选了iptable，是因为选dnsmasq-full的时候勾选了iptable支持，所以强行装上了iptable

恰好我遇到了这个问题，openwrt用nftable，但是全锥NAT自动调用iptables，也是因为安装了dnsmasq-full才有了iptables，现在我是不是应该卸载iptables（没有规则，所有规则都在nftable）

chenmozhijin, 2024-02-12:
> 恰好我遇到了这个问题，openwrt用nftable，但是全锥NAT自动调用iptables，也是因为安装了dnsmasq-full才有了iptables，现在我是不是应该卸载iptables（没有规则，所有规则都在nftable）

能发一下编译的配置与系统日志吗，另外为什么您认为全锥NAT自动调用iptables呢？
我看这个错误与https://github.com/chenmozhijin/turboacc/issues/8 很像

mis1042, 2024-02-12:
> 能发一下编译的配置与系统日志吗，另外为什么您认为全锥NAT自动调用iptables呢？
> 我看这个错误与https://github.com/chenmozhijin/turboacc/issues/8#issuecomment-1663908360 很像

排查了一下全锥NAT打不开的问题，这个是因为zone的配置文件错误导致的，把firewall配置文件里的fullcone改为fullcone4即可
不过断流的问题我也遇到了，正在排查

Azreulo, 2024-02-19:
没想到这个Bug竟然还没被修复。

这个Bug貌似只会出现在使用nginx时，原因是有个空格，具体去lean的Issus里面翻翻。

解决办法：
编译前：
`sed -i 's|status fastpath|status fastpath >/dev/null|g' package/turboacc/luci-app-turboacc/luasrc/controller/turboacc.lua`
路由器上：
`sed -i 's|status fastpath|status fastpath >/dev/null|g' /usr/lib/lua/luci/controller/turboacc.lua`
`rm -rf /tmp/luci-*`

然后刷新下页面就可以了。