chenmozhijin/turboacc #20: 无法启用Fullcone NAT

sht2017, 2024-02-24:
[image]
如上
```
# find /lib/modules/ | grep nft_fullcone.ko
/lib/modules/5.15.137/nft_fullcone.ko
# /etc/init.d/firewall restart
Section @defaults[0] specifies unknown option 'fullcone'
# cat "/sys/module/nft_flow_offload/refcnt"
1
# cat "/proc/sys/net/ipv4/tcp_congestion_control"
bbr
# cat "/sys/module/nft_fullcone/refcnt"
0
```

chenmozhijin, 2024-02-24:
应该是[firewall的补丁](https://github.com/chenmozhijin/turboacc/tree/package/firewall4-698a53354fd280aae097efe08803c0c9a10c14c2/firewall4/patches)没打上

sht2017, 2024-02-24:
有点像 #18，但是是存在nft_fullcone.ko的

chenmozhijin, 2024-02-24:
[libnftnl](https://github.com/chenmozhijin/turboacc/tree/package/libnftnl-1.2.6/libnftnl/patches)与[nftables](https://github.com/chenmozhijin/turboacc/tree/package/nftables-1.0.9/nftables/patches)的补丁打了吗?

我运行
```
/usr/sbin/nft -c 'add table inet fw4-fullcone-test; add chain inet fw4-fullcone-test dstnat { type nat hook prerouting priority -100; policy
accept; fullcone; }; add chain inet fw4-fullcone-test srcnat { type nat hook postrouting priority -100; policy accept; fullcone; }; '
```
是没有任何输出的

sht2017, 2024-02-24:
> ```shell
> /etc/init.d/firewall restart
> ```

patch以后
`/usr/sbin/nft -c 'add table inet fw4-fullcone-test; add chain inet fw4-fullcone-test dstnat { type nat hook prerouting priority -100; policy accept; fullcone; }; add chain inet fw4-fullcone-test srcnat { type nat hook postrouting priority -100; policy accept; fullcone; }; '` 和 `/etc/init.d/firewall restart`的输出都没了，但还是启用不了：
```shell
# find /lib/modules/ | grep nft_fullcone.ko
/lib/modules/5.15.137/nft_fullcone.ko
# /etc/init.d/firewall restart
# cat "/sys/module/nft_flow_offload/refcnt"
1
# cat "/proc/sys/net/ipv4/tcp_congestion_control"
bbr
# cat "/sys/module/nft_fullcone/refcnt"
0
```

chenmozhijin, 2024-02-24:
`cat /etc/config/firewall`看一下防火墙配置里面defaults下有没有`option fullcone '1'`，wan的zone下有没有`option fullcone4 '1'`

sht2017, 2024-02-24:
> `cat /etc/config/firewall`看一下防火墙配置里面defaults下有没有`option fullcone '1'`，wan的zone下有没有`option fullcone4 '1'`

```
# cat /etc/config/firewall

config defaults
        option syn_flood '1'
        option input 'REJECT'
        option output 'ACCEPT'
        option forward 'REJECT'
        option flow_offloading '1'
        option flow_offloading_hw '1'
        option fullcone '1'

config zone
        option name 'lan'
        option input 'ACCEPT'
        option output 'ACCEPT'
        option forward 'ACCEPT'
        list network 'lan'

config zone
        option name 'wan'
        option input 'REJECT'
        option output 'ACCEPT'
        option forward 'REJECT'
        option masq '1'
        option mtu_fix '1'
        list network 'wan'
        list network 'wan6'
        list network 'trm_wwan'
        list network 'trm_wwan6'

config forwarding
        option src 'lan'
        option dest 'wan'

config rule
        option name 'Allow-DHCP-Renew'
        option src 'wan'
        option proto 'udp'
        option dest_port '68'
        option target 'ACCEPT'
        option family 'ipv4'

config rule
        option name 'Allow-Ping'
        option src 'wan'
        option proto 'icmp'
        option icmp_type 'echo-request'
        option family 'ipv4'
        option target 'ACCEPT'

config rule
        option name 'Allow-IGMP'
        option src 'wan'
        option proto 'igmp'
        option family 'ipv4'
        option target 'ACCEPT'

config rule
        option name 'Allow-DHCPv6'
        option src 'wan'
        option proto 'udp'
        option dest_port '546'
        option family 'ipv6'
        option target 'ACCEPT'

config rule
        option name 'Allow-MLD'
        option src 'wan'
        option proto 'icmp'
        option src_ip 'fe80::/10'
        list icmp_type '130/0'
        list icmp_type '131/0'
        list icmp_type '132/0'
        list icmp_type '143/0'
        option family 'ipv6'
        option target 'ACCEPT'

config rule
        option name 'Allow-ICMPv6-Input'
        option src 'wan'
        option proto 'icmp'
        list icmp_type 'echo-request'
        list icmp_type 'echo-reply'
        list icmp_type 'destination-unreachable'
        list icmp_type 'packet-too-big'
        list icmp_type 'time-exceeded'
        list icmp_type 'bad-header'
        list icmp_type 'unknown-header-type'
        list icmp_type 'router-solicitation'
        list icmp_type 'neighbour-solicitation'
        list icmp_type 'router-advertisement'
        list icmp_type 'neighbour-advertisement'
        option limit '1000/sec'
        option family 'ipv6'
        option target 'ACCEPT'

config rule
        option name 'Allow-ICMPv6-Forward'
        option src 'wan'
        option dest '*'
        option proto 'icmp'
        list icmp_type 'echo-request'
        list icmp_type 'echo-reply'
        list icmp_type 'destination-unreachable'
        list icmp_type 'packet-too-big'
        list icmp_type 'time-exceeded'
        list icmp_type 'bad-header'
        list icmp_type 'unknown-header-type'
        option limit '1000/sec'
        option family 'ipv6'
        option target 'ACCEPT'

config rule
        option name 'Allow-IPSec-ESP'
        option src 'wan'
        option dest 'lan'
        option proto 'esp'
        option target 'ACCEPT'

config rule
        option name 'Allow-ISAKMP'
        option src 'wan'
        option dest 'lan'
        option dest_port '500'
        option proto 'udp'
        option target 'ACCEPT'
```

closed 2024-02-26 by sht2017

King-Of-Knights, 2024-04-18:
@chenmozhijin 大佬好，按照上述步骤以后，用`/etc/init.d/firewall restart`重启防火墙后
我看到两个相关信息：
```
Section @defaults[0] specifies unknown option 'fullcone6'
Section @zone[1] (wan) IPv4 fullcone enabled for zone 'wan'
```
但是在luci界面里面的全锥型NAT开关失效了

chenmozhijin, 2024-04-18:
能发一下/etc/config/firewall的防火墙配置吗？

King-Of-Knights, 2024-04-18:
@chenmozhijin 翻了一下其他问题，好像有说自带的ft-fullcone会造成问题，我先试试删了
```
config defaults
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option flow_offloading '1'
	option fullcone '1'
	option fullcone6 '0'
	option flow_offloading_hw '1'
	option synflood_protect '1'

config zone
	option name 'lan'
	list network 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'

config zone
	option name 'wan'
	list network 'wan'
	list network 'wan6'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option masq '1'
	option mtu_fix '1'
	option fullcone4 '1'

config forwarding
	option src 'lan'
	option dest 'wan'

config rule
	option name 'Allow-DHCP-Renew'
	option src 'wan'
	option proto 'udp'
	option dest_port '68'
	option target 'ACCEPT'
	option family 'ipv4'

config rule
	option name 'Allow-Ping'
	option src 'wan'
	option proto 'icmp'
	option icmp_type 'echo-request'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-IGMP'
	option src 'wan'
	option proto 'igmp'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-DHCPv6'
	option src 'wan'
	option proto 'udp'
	option dest_port '546'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-MLD'
	option src 'wan'
	option proto 'icmp'
	option src_ip 'fe80::/10'
	list icmp_type '130/0'
	list icmp_type '131/0'
	list icmp_type '132/0'
	list icmp_type '143/0'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Input'
	option src 'wan'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	list icmp_type 'router-solicitation'
	list icmp_type 'neighbour-solicitation'
	list icmp_type 'router-advertisement'
	list icmp_type 'neighbour-advertisement'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Forward'
	option src 'wan'
	option dest '*'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-IPSec-ESP'
	option src 'wan'
	option dest 'lan'
	option proto 'esp'
	option target 'ACCEPT'

config rule
	option name 'Allow-ISAKMP'
	option src 'wan'
	option dest 'lan'
	option dest_port '500'
	option proto 'udp'
	option target 'ACCEPT'

config zone 'ipsecserver'
	option name 'ipsecserver'
	option input 'ACCEPT'
	option forward 'ACCEPT'
	option output 'ACCEPT'
	option network 'ipsec_server'

config include 'luci_app_ipsec_server'
	option type 'script'
	option path '/var/etc/ipsecvpn.include'
	option reload '1'

config include 'passwall'
	option type 'script'
	option path '/var/etc/passwall.include'
	option reload '1'

config include 'passwall_server'
	option type 'script'
	option path '/var/etc/passwall_server.include'
	option reload '1'

config include 'iptvhelper'
	option type 'script'
	option path '/etc/firewall.iptvhelper'
	option family 'any'
	option reload '1'
```

chenmozhijin, 2024-04-18:
把wan下的`	option fullcone4 '1'`与defaults下的`	option fullcone6 '0'`删掉应该就好了，像这样：
```
config defaults
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option flow_offloading '1'
	option fullcone '1'
	option flow_offloading_hw '1'
	option synflood_protect '1'

config zone
	option name 'lan'
	list network 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'

config zone
	option name 'wan'
	list network 'wan'
	list network 'wan6'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option masq '1'
	option mtu_fix '1'

config forwarding
	option src 'lan'
	option dest 'wan'

config rule
	option name 'Allow-DHCP-Renew'
	option src 'wan'
	option proto 'udp'
	option dest_port '68'
	option target 'ACCEPT'
	option family 'ipv4'

config rule
	option name 'Allow-Ping'
	option src 'wan'
	option proto 'icmp'
	option icmp_type 'echo-request'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-IGMP'
	option src 'wan'
	option proto 'igmp'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-DHCPv6'
	option src 'wan'
	option proto 'udp'
	option dest_port '546'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-MLD'
	option src 'wan'
	option proto 'icmp'
	option src_ip 'fe80::/10'
	list icmp_type '130/0'
	list icmp_type '131/0'
	list icmp_type '132/0'
	list icmp_type '143/0'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Input'
	option src 'wan'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	list icmp_type 'router-solicitation'
	list icmp_type 'neighbour-solicitation'
	list icmp_type 'router-advertisement'
	list icmp_type 'neighbour-advertisement'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Forward'
	option src 'wan'
	option dest '*'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-IPSec-ESP'
	option src 'wan'
	option dest 'lan'
	option proto 'esp'
	option target 'ACCEPT'

config rule
	option name 'Allow-ISAKMP'
	option src 'wan'
	option dest 'lan'
	option dest_port '500'
	option proto 'udp'
	option target 'ACCEPT'

config zone 'ipsecserver'
	option name 'ipsecserver'
	option input 'ACCEPT'
	option forward 'ACCEPT'
	option output 'ACCEPT'
	option network 'ipsec_server'

config include 'luci_app_ipsec_server'
	option type 'script'
	option path '/var/etc/ipsecvpn.include'
	option reload '1'

config include 'passwall'
	option type 'script'
	option path '/var/etc/passwall.include'
	option reload '1'

config include 'passwall_server'
	option type 'script'
	option path '/var/etc/passwall_server.include'
	option reload '1'

config include 'iptvhelper'
	option type 'script'
	option path '/etc/firewall.iptvhelper'
	option family 'any'
	option reload '1'
```

King-Of-Knights, 2024-04-18:
@chenmozhijin 感谢佬的深夜回复啊，我刚才重新编译了一下，然后清空设置以后重新刷入了，感觉好像正确多了，也没有出现`Section @defaults[0] specifies unknown option 'fullcone6'`了，感觉这个是没有清空以前防火墙配置的锅，但好像luci里面的开关还是不能设定`option fullcone4 '1’`和 `option fullcone6 '0'`，希望佬有时间能修修！
```
config defaults
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option fullcone '1'
	option brcmfullcone '0'
	option flow_offloading '1'
	option flow_offloading_hw '1'
	option synflood_protect '1'

config zone
	option name 'lan'
	list network 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'

config zone
	option name 'wan'
	list network 'wan'
	list network 'wan6'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option fullcone4 '1'
	option fullcone6 '0'
	option masq '1'
	option mtu_fix '1'
```

King-Of-Knights, 2024-04-20:
@chenmozhijin 大佬好呀，经过一晚上的研究，发现是最近官方对luci-base频繁更新，导致该插件fullcone开关出现问题，我退回2月10日的luci，fullcone开关就正常了