chenmozhijin/turboacc #23: 手动添加openwrt失败，提示 -ash: bash: not found

lzwdavid, 2024-04-08:
# curl -ssl https://raw.githubusercontent.com/chenmozhijin/tu
rboacc/luci/add_turboacc.sh -o add_turboacc.sh && bash add_turboacc.sh
-ash: bash: not found
root@Netgear:~# pwd
/root
root@Netgear:~# cd /
root@Netgear:/# curl -ssl https://raw.githubusercontent.com/chenmozhijin/tu
rboacc/luci/add_turboacc.sh -o add_turboacc.sh && bash add_turboacc.sh
-ash: bash: not found
求问如何解决？谢谢。

lzwdavid, 2024-04-08:
补充：
OpenWrt 22.03.5 r20134-5f15225c1e / LuCI openwrt-22.03 branch git-23.119.80898-65ef406
Kernel Version | 5.10.176

chenmozhijin, 2024-04-08:
要在openwrt的源码目录中运行这行命令。

lzwdavid, 2024-04-08:
> 要在openwrt的源码目录中运行这行命令。

感谢
我已经在./package 目录中试过了，仍然说 ./package not found.

chenmozhijin, 2024-04-08:
是克隆了openwrt源码的存储库的那个目录。里面包含package 这个文件夹。就是git clone https://git.openwrt.org/openwrt/openwrt.git再cd openwrt的目录

closed 2024-04-11 by chenmozhijin

lzwdavid, 2024-04-14:
问过一圈，有人说 “[badcrazy 发表于 2024-4-8 12:13](https://www.right.com.cn/forum/forum.php?mod=redirect&goto=findpost&pid=20138174&ptid=8363121)
turboacc只能编译进固件，事后安装装不进去的”