chenmozhijin/turboacc #24: 无fullcone的选项

OldCoding, 2024-05-19:
[image]
用的这条命令curl -sSL https://raw.githubusercontent.com/chenmozhijin/turboacc/luci/add_turboacc.sh -o add_turboacc.sh && bash add_turboacc.sh
下面是命令执行过程

+ trap 'rm -rf "$TMPDIR"' EXIT
++ mktemp -d
+ TMPDIR=/tmp/tmp.9o03hu6wdf
+ '[' -d ./package ']'
++ sed -n '/VERSION_NUMBER:=$(if $(VERSION_NUMBER),$(VERSION_NUMBER),.*)/p' include/version.mk
++ sed -e 's/.*$(VERSION_NUMBER),//' -e 's/)//'
+ VERSION_NUMBER=SNAPSHOT
++ find ./include
++ sed -n '/kernel-[0-9]/p'
++ sed ':a;N;$!ba;s/\n/ /g'
++ sed -e s@./include/kernel-@@
+ kernel_versions='6.6 6.1 5.15'
+ '[' -z '6.6 6.1 5.15' ']'
+ echo 'kernel version: 6.6 6.1 5.15'
+ '[' -d ./package/turboacc ']'
kernel version: 6.6 6.1 5.15
+ git clone --depth=1 --single-branch https://github.com/fullcone-nat-nftables/nft-fullcone /tmp/tmp.9o03hu6wdf/turboacc/nft-fullcone
Cloning into '/tmp/tmp.9o03hu6wdf/turboacc/nft-fullcone'...
+ git clone --depth=1 --single-branch https://github.com/chenmozhijin/turboacc /tmp/tmp.9o03hu6wdf/turboacc/turboacc
Cloning into '/tmp/tmp.9o03hu6wdf/turboacc/turboacc'...
+ [[ 0 = 2 ]]
+ git clone --depth=1 --single-branch --branch package https://github.com/chenmozhijin/turboacc /tmp/tmp.9o03hu6wdf/package
Cloning into '/tmp/tmp.9o03hu6wdf/package'...
+ cp -r /tmp/tmp.9o03hu6wdf/turboacc/turboacc/luci-app-turboacc /tmp/tmp.9o03hu6wdf/turboacc/luci-app-turboacc
+ rm -rf /tmp/tmp.9o03hu6wdf/turboacc/turboacc
+ cp -r /tmp/tmp.9o03hu6wdf/package/shortcut-fe /tmp/tmp.9o03hu6wdf/turboacc/shortcut-fe
+ for kernel_version in $kernel_versions
+ patch_953_path=./target/linux/generic/hack-6.6/953-net-patch-linux-kernel-to-support-shortcut-fe.patch
+ patch_613_path=./target/linux/generic/pending-6.6/613-netfilter_optional_tcp_window_check.patch
+ '[' 6.6 = 6.6 ']'
+ patch_9[52](https://github.com/OldCoding/openwrt_packit_arm/actions/runs/9145051682/job/25143578837#step:7:53)_path=./target/linux/generic/hack-6.6/952-add-net-conntrack-events-support-multiple-registrant.patch
+ patch_952=952-add-net-conntrack-events-support-multiple-registrant.patch
+ for file_path in "$patch_952_path" "$patch_9[53](https://github.com/OldCoding/openwrt_packit_arm/actions/runs/9145051682/job/25143578837#step:7:54)_path" "$patch_613_path"
+ '[' -a ./target/linux/generic/hack-6.6/952-add-net-conntrack-events-support-multiple-registrant.patch ']'
+ for file_path in "$patch_952_path" "$patch_953_path" "$patch_613_path"
+ '[' -a ./target/linux/generic/hack-6.6/953-net-patch-linux-kernel-to-support-shortcut-fe.patch ']'
+ for file_path in "$patch_952_path" "$patch_953_path" "$patch_613_path"
+ '[' -a ./target/linux/generic/pending-6.6/613-netfilter_optional_tcp_window_check.patch ']'
+ cp -f /tmp/tmp.9o03hu6wdf/package/hack-6.6/952-add-net-conntrack-events-support-multiple-registrant.patch ./target/linux/generic/hack-6.6/952-add-net-conntrack-events-support-multiple-registrant.patch
+ cp -f /tmp/tmp.9o03hu6wdf/package/hack-6.6/953-net-patch-linux-kernel-to-support-shortcut-fe.patch ./target/linux/generic/hack-6.6/953-net-patch-linux-kernel-to-support-shortcut-fe.patch
+ cp -f /tmp/tmp.9o03hu6wdf/package/pending-6.6/613-netfilter_optional_tcp_window_check.patch ./target/linux/generic/pending-6.6/613-netfilter_optional_tcp_window_check.patch
+ grep -q CONFIG_NF_CONNTRACK_CHAIN_EVENTS ./target/linux/generic/config-6.6
+ echo '# CONFIG_NF_CONNTRACK_CHAIN_EVENTS is not set'
+ grep -q CONFIG_SHORTCUT_FE ./target/linux/generic/config-6.6
+ echo '# CONFIG_SHORTCUT_FE is not set'
+ for kernel_version in $kernel_versions
+ patch_953_path=./target/linux/generic/hack-6.1/953-net-patch-linux-kernel-to-support-shortcut-fe.patch
+ patch_613_path=./target/linux/generic/pending-6.1/613-netfilter_optional_tcp_window_check.patch
+ '[' 6.1 = 6.6 ']'
+ '[' 6.1 = 6.1 ']'
+ patch_952_path=./target/linux/generic/hack-6.1/952-add-net-conntrack-events-support-multiple-registrant.patch
+ patch_952=952-add-net-conntrack-events-support-multiple-registrant.patch
+ for file_path in "$patch_952_path" "$patch_953_path" "$patch_613_path"
+ '[' -a ./target/linux/generic/hack-6.1/952-add-net-conntrack-events-support-multiple-registrant.patch ']'
+ for file_path in "$patch_952_path" "$patch_953_path" "$patch_613_path"
+ '[' -a ./target/linux/generic/hack-6.1/953-net-patch-linux-kernel-to-support-shortcut-fe.patch ']'
+ for file_path in "$patch_952_path" "$patch_953_path" "$patch_613_path"
+ '[' -a ./target/linux/generic/pending-6.1/613-netfilter_optional_tcp_window_check.patch ']'
+ cp -f /tmp/tmp.9o03hu6wdf/package/hack-6.1/952-add-net-conntrack-events-support-multiple-registrant.patch ./target/linux/generic/hack-6.1/952-add-net-conntrack-events-support-multiple-registrant.patch
+ cp -f /tmp/tmp.9o03hu6wdf/package/hack-6.1/953-net-patch-linux-kernel-to-support-shortcut-fe.patch ./target/linux/generic/hack-6.1/953-net-patch-linux-kernel-to-support-shortcut-fe.patch
+ cp -f /tmp/tmp.9o03hu6wdf/package/pending-6.1/613-netfilter_optional_tcp_window_check.patch ./target/linux/generic/pending-6.1/613-netfilter_optional_tcp_window_check.patch
+ grep -q CONFIG_NF_CONNTRACK_CHAIN_EVENTS ./target/linux/generic/config-6.1
+ echo '# CONFIG_NF_CONNTRACK_CHAIN_EVENTS is not set'
+ grep -q CONFIG_SHORTCUT_FE ./target/linux/generic/config-6.1
+ echo '# CONFIG_SHORTCUT_FE is not set'
+ for kernel_version in $kernel_versions
+ patch_953_path=./target/linux/generic/hack-5.15/953-net-patch-linux-kernel-to-support-shortcut-fe.patch
+ patch_613_path=./target/linux/generic/pending-5.15/613-netfilter_optional_tcp_window_check.patch
+ '[' 5.15 = 6.6 ']'
+ '[' 5.15 = 6.1 ']'
+ '[' 5.15 = 5.15 ']'
+ patch_952_path=./target/linux/generic/hack-5.15/952-add-net-conntrack-events-support-multiple-registrant.patch
+ patch_952=952-add-net-conntrack-events-support-multiple-registrant.patch
+ for file_path in "$patch_952_path" "$patch_953_path" "$patch_613_path"
+ '[' -a ./target/linux/generic/hack-5.15/952-add-net-conntrack-events-support-multiple-registrant.patch ']'
+ for file_path in "$patch_952_path" "$patch_953_path" "$patch_613_path"
+ '[' -a ./target/linux/generic/hack-5.15/953-net-patch-linux-kernel-to-support-shortcut-fe.patch ']'
+ for file_path in "$patch_952_path" "$patch_953_path" "$patch_613_path"
+ '[' -a ./target/linux/generic/pending-5.15/613-netfilter_optional_tcp_window_check.patch ']'
+ cp -f /tmp/tmp.9o03hu6wdf/package/hack-5.15/952-add-net-conntrack-events-support-multiple-registrant.patch ./target/linux/generic/hack-5.15/952-add-net-conntrack-events-support-multiple-registrant.patch
+ cp -f /tmp/tmp.9o03hu6wdf/package/hack-5.15/953-net-patch-linux-kernel-to-support-shortcut-fe.patch ./target/linux/generic/hack-5.15/953-net-patch-linux-kernel-to-support-shortcut-fe.patch
+ cp -f /tmp/tmp.9o03hu6wdf/package/pending-5.15/613-netfilter_optional_tcp_window_check.patch ./target/linux/generic/pending-5.15/613-netfilter_optional_tcp_window_check.patch
+ grep -q CONFIG_NF_CONNTRACK_CHAIN_EVENTS ./target/linux/generic/config-5.15
+ echo '# CONFIG_NF_CONNTRACK_CHAIN_EVENTS is not set'
+ grep -q CONFIG_SHORTCUT_FE ./target/linux/generic/config-5.15
+ echo '# CONFIG_SHORTCUT_FE is not set'
+ cp -r /tmp/tmp.9o03hu6wdf/turboacc ./package/turboacc
+ rm -rf ./package/libs/libnftnl ./package/network/config/firewall4 ./package/network/utils/nftables
+ [[ SNAPSHOT =~ ^22.03.* ]]
++ grep -o 'FIREWALL4_VERSION=.*' /tmp/tmp.9o03hu6wdf/package/version
++ cut -d = -f 2
+ FIREWALL4_VERSION=698a533[54](https://github.com/OldCoding/openwrt_packit_arm/actions/runs/9145051682/job/25143578837#step:7:55)fd280aae097efe08803c0c9a10c14c2
++ grep -o 'LIBNFTNL_VERSION=.*' /tmp/tmp.9o03hu6wdf/package/version
++ cut -d = -f 2
+ LIBNFTNL_VERSION=1.2.6
++ grep -o 'NFTABLES_VERSION=.*' /tmp/tmp.9o03hu6wdf/package/version
++ cut -d = -f 2
+ NFTABLES_VERSION=1.0.9
+ cp -RT /tmp/tmp.9o03hu6wdf/package/firewall4-698a53354fd280aae097efe08803c0c9a10c14c2/firewall4 ./package/network/config/firewall4
+ cp -RT /tmp/tmp.9o03hu6wdf/package/libnftnl-1.2.6/libnftnl ./package/libs/libnftnl
+ cp -RT /tmp/tmp.9o03hu6wdf/package/nftables-1.0.9/nftables ./package/network/utils/nftables
+ echo Finish
Finish
+ exit 0
+ rm -rf /tmp/tmp.9o03hu6wdf
使用的immortalwrt源码，master分支，编译的时候已经把/package/network/utils/fullconenat-nft文件删除
编译出来还是没有fullcone选项，
包括官方opemwrt源码也一样没有这个选项

chenmozhijin, 2024-05-19:
源码的package/turboacc文件夹中是否有nft-fullcone文件夹
make menuconfig后> Kernel modules > Netfilter Extensions下是否有kmod-nft-fullcone是否勾选
编译的固件中是否包含kmod-nft-fullcone软件包
固件中是否有/lib/modules/内核版本/nft_fullcone.ko文件

OldCoding, 2024-05-19:
> kmod-nft-fullcone
package/turboacc文件夹中有三个文件夹
luci-app-turboacc、nft-fullcone、shortcut-fe
kmod-nft-fullcone有勾选，编译出来的固件也有kmod-nft-fullcone软件包
/lib/modules/内核版本下面没有nft_fullcone.ko文件
内核是f大的6.6内核，设备是N1

OldCoding, 2024-05-19:
kmod-nft-fullcone编译出来的版本是6.1.89-r1，这个有影响嘛

OldCoding, 2024-05-20:
> > > kmod-nft-fullcone
> > 
> > 
> > package/turboacc文件夹中有三个文件夹 luci-app-turboacc、nft-fullcone、shortcut-fe kmod-nft-fullcone有勾选，编译出来的固件也有kmod-nft-fullcone软件包 /lib/modules/内核版本下面没有nft_fullcone.ko文件 内核是f大的6.6内核，设备是N1
> > f大的内核?

内核作者flippy，设备是arm盒子，斐讯n1，内核仓库https://github.com/unifreq/linux-6.6.y

chenmozhijin, 2024-05-20:
nft_fullcone.ko是内核模块,可能是换内核的时候把它换没了,可以用原来的内核试试看。

OldCoding, 2024-05-20:
> nft_fullcone.ko是内核模块,可能是换内核的时候把它换没了,可以用原来的内核试试看。

了解了，是这个内核没有这个模块，需要其他方法补进去

closed 2024-05-20 by OldCoding