chenmozhijin/turboacc #25: ax6 immortalwrt 23.05 编译出错

xkcore123, 2024-05-31:
删除了953 613补丁
编译错误提示如下
patching file include/uapi/linux/in.h
patching file tools/include/uapi/linux/in.h
patching file net/netfilter/nf_conntrack_ecache.c
Hunk #1 FAILED at 266.
1 out of 1 hunk FAILED -- saving rejects to file net/netfilter/nf_conntrack_ecache.c.rej
patching file include/net/netns/conntrack.h
patching file net/netfilter/nf_conntrack_proto_tcp.c
patching file net/netfilter/nf_conntrack_standalone.c
Patch failed!  Please fix /home/core/ax6-immortalwrt/target/linux/qualcommax/patches-6.6/0600-1-qca-nss-ecm-support-CORE.patch!
make[4]: *** [Makefile:32: /home/core/ax6-immortalwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.6.30/.prepared_1e42c204ee5948ef57272df366bdfacb] Error 1
make[4]: Leaving directory '/home/core/ax6-immortalwrt/target/linux/qualcommax'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/core/ax6-immortalwrt/target/linux'
time: target/linux/compile#2.45#10.23#9.27
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:32: target/linux/compile] Error 1
make[2]: Leaving directory '/home/core/ax6-immortalwrt'
make[1]: *** [target/Makefile:25: /home/core/ax6-immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/core/ax6-immortalwrt'
大佬，请问这是啥原因
make: *** [/home/core/ax6-immortalwrt/include/toplevel.mk:233: world] Error 2
core@ubuntu:~/ax6-immortalwrt$ make menuconfig
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libpcre', which does not exist
tmp/.config-package.in:61501:error: recursive dependency detected!
tmp/.config-package.in:61501:   symbol PACKAGE_luci-app-turboacc depends on PACKAGE_kmod-nft-fullcone
tmp/.config-package.in:15370:   symbol PACKAGE_kmod-nft-fullcone is selected by PACKAGE_luci-app-turboacc
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

chenmozhijin, 2024-05-31:
这是哪里的immortalwrt 23.05，为什么是6.6内核？
immortalwrt的openwrt-23.05与master分支都没有0600-1-qca-nss-ecm-support-CORE.patch这个补丁
这应该是0600-1-qca-nss-ecm-support-CORE.patch这个补丁的问题可能它与952或613补丁不兼容

xkcore123, 2024-05-31:
> 这是哪里的immortalwrt 23.05，为什么是6.6内核？ immortalwrt的openwrt-23.05与master分支都没有0600-1-qca-nss-ecm-support-CORE.patch这个补丁 这应该是0600-1-qca-nss-ecm-support-CORE.patch这个补丁的问题可能它与952或613补丁不兼容

我用的这个https://github.com/DoveKi/immortalwrt，现在换个试试，谢谢

closed 2024-05-31 by xkcore123