chenmozhijin/turboacc #27: 如不需要sfe可以删除953与613补丁。请问怎么删除呢

zbuhui, 2024-08-16:
如题，可以详细说说吗，谢谢

chenmozhijin, 2024-08-17:
删除
target/linux/generic/hack-$kernel_version/953-net-patch-linux-kernel-to-support-shortcut-fe.patch
target/linux/generic/pending-$kernel_version/613-netfilter_optional_tcp_window_check.patch

$kernel_version为openwrt的内核版本

chenmozhijin, 2024-08-19:
lede?lede本身就有turboacc吧

zbuhui, 2024-08-20:
是的 我是硬路由刷入自己编译的lede无限自动重启，找的这个固件不支持nat1，所以想重新编译了替代这个固件默认的看看会不会有效

chenmozhijin, 2024-08-20:
lede现在[默认编译](https://github.com/coolsnowwolf/lede/blob/5ae14cb43d1711565bf08a6916d6d39ca0da64a8/include/target.mk#L56)的是firewall3不是firewall4,这些firewall4、libnftnl、nftables的补丁也是适配firewall4用的。你编译的这个firewall应该是firewall3它后端调用的是Iptables不是nftables。lede无限自动重启可以试着在lede的[issues](https://github.com/coolsnowwolf/lede/issues)提问,或许会有回复。

我现在使用的是用openwrt官方v23.05.4版本源码编译的,感兴趣的话可以试试[OpenWrt-K](https://github.com/chenmozhijin/OpenWrt-K)进行云编译

zbuhui, 2024-08-21:
非常感谢，我试试看

closed 2024-08-24 by chenmozhijin