chenmozhijin/turboacc #28: OpenWrt 官方版 openwrt-23.05.4 确实很多依赖

Gloryandel, 2024-08-24:
Arch：x86_64
```
Ubuntu22:~/openwrt-23.05.4$ ./scripts/feeds install -a
WARNING: Makefile 'package/app/turboacc/luci-app-turboacc/Makefile' has a dependency on 'kmod-fast-classifier', which does not exist
WARNING: Makefile 'package/app/turboacc/luci-app-turboacc/Makefile' has a dependency on 'kmod-shortcut-fe-cm', which does not exist
WARNING: Makefile 'package/app/turboacc/luci-app-turboacc/Makefile' has a dependency on 'kmod-shortcut-fe-drv', which does not exist
WARNING: Makefile 'package/app/turboacc/luci-app-turboacc/Makefile' has a dependency on 'kmod-nft-fullcone', which does not exist
```

chenmozhijin, 2024-08-24:
你是怎么安装的turboacc的，是使用方法中的命令吗？

Gloryandel, 2024-08-24:
> 你是怎么安装的turboacc的，是使用方法中的命令吗？

git 克隆进 package/app/，奇怪的是很早之前的 23.05.3 需要的模块均有。同路径下的 oppenclash 可以正常编译。

chenmozhijin, 2024-08-24:
需要在openwrt源代码目录执行：
```
curl -sSL https://raw.githubusercontent.com/chenmozhijin/turboacc/luci/add_turboacc.sh -o add_turboacc.sh && bash add_turboacc.sh
```
会自动安装依赖

Gloryandel, 2024-08-24:
> 需要在openwrt源代码目录执行：
> 
> ```
> curl -sSL https://raw.githubusercontent.com/chenmozhijin/turboacc/luci/add_turboacc.sh -o add_turboacc.sh && bash add_turboacc.sh
> ```
> 
> 会自动安装依赖

感谢大佬指点，犯了经验错了😳

chenmozhijin, 2024-08-24:
记得把原来你clone删掉

closed 2024-08-24 by chenmozhijin