chenmozhijin/turboacc #29: flippy编译的内核，系统是openwrt官方的主线

jsdfhasuh, 2024-09-12:
/usr/sbin/nft -c 'add table inet fw4-fullcone-test; add chain inet fw4-fullcone-test dstnat { type nat hook prerouting priority -100; policy accept; fullcone; }; add chain inet fw4-fullcone-test srcnat { type nat hook postrouting priority -100; policy accept; fullcone; }; '
Error: No such file or directory; did you mean table ‘fw4’ in family inet?
add table inet fw4-fullcone-test; add chain inet fw4-fullcone-test dstnat { type nat hook prerouting priority -100; policy accept; fullcone; }; add chain inet fw4-fullcone-test srcnat { type nat hook postrouting priority -100; policy accept; fullcone; }; 
                                                 ^^^^^^^^^^^^^^^^^
Error: No such file or directory; did you mean table ‘fw4’ in family inet?
add table inet fw4-fullcone-test; add chain inet fw4-fullcone-test dstnat { type nat hook prerouting priority -100; policy accept; fullcone; }; add chain inet fw4-fullcone-test srcnat { type nat hook postrouting priority -100; policy accept; fullcone; };

chenmozhijin, 2024-09-12:
看着像没打firewall4的补丁，你是怎么添加的turboacc的？

jsdfhasuh, 2024-09-12:
内核模块没有成功加载，请问应该如何在别人的内核下面打补丁，这个不太懂怎么操作

jsdfhasuh, 2024-09-12:
我拿到内核的源码，但是就是不知道打补丁上去

chenmozhijin, 2024-09-12:
是内核的,还是openwrt的？

jsdfhasuh, 2024-09-13:
> 是内核的,还是openwrt的？ 目录结构是什么样的？
> 
> 除了内核firewall4、libnftnl、nftables这几个包也需要打补丁的

1.是Linux内核的。是在官方openwrt主线下面编译了系统，然后再用自定义的内核更换了openwrt的内核，然后再安装上去
2.在自定义linux内核的编译menuconfig里面，没有找到nft-fullcone 应该是需要打包
3.源码目录https://github.com/unifreq/linux-6.6.y

chenmozhijin, 2024-09-13:
试一下在内核源码目录执行
`curl -s -L --retry 6 https://github.com/chenmozhijin/turboacc/raw/package/hack-6.6/952-add-net-conntrack-events-support-multiple-registrant.patch  | patch -p1 -d .`
然后在内核配置设置CONFIG_NFT_FULLCONE=y CONFIG_NF_NAT=y CONFIG_NF_NAT_IPV6=y CONFIG_NF_CONNTRACK_EVENTS=y CONFIG_NF_CONNTRACK_CHAIN_EVENTS=y

jsdfhasuh, 2024-09-13:
Hunk #1 FAILED at 65.
Hunk #2 FAILED at 98.
Hunk #3 FAILED at 117.
3 out of 3 hunks FAILED -- saving rejects to file include/net/netfilter/nf_conntrack_ecache.h.rej
patching file include/net/netns/conntrack.h
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file include/net/netns/conntrack.h.rej
patching file net/netfilter/Kconfig
Hunk #1 succeeded at 172 (offset 11 lines).
patching file net/netfilter/nf_conntrack_core.c
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file net/netfilter/nf_conntrack_core.c.rej
patching file net/netfilter/nf_conntrack_ecache.c
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Skipping patch.
7 out of 7 hunks ignored -- saving rejects to file net/netfilter/nf_conntrack_ecache.c.rej
patching file net/netfilter/nf_conntrack_netlink.c
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Skipping patch.
5 out of 5 hunks ignored -- saving rejects to file net/netfilter/nf_conntrack_netlink.c.re
好像是已经打过补丁了

chenmozhijin, 2024-09-13:
firewall4、libnftnl、nftables打了补丁了吗？

jsdfhasuh, 2024-09-13:
> firewall4、libnftnl、nftables打了补丁了吗？

这怎么看有没有打上补丁，我发现部分补丁没有正常打上

jsdfhasuh, 2024-09-13:
--- include/net/netfilter/nf_conntrack_ecache.h
+++ include/net/netfilter/nf_conntrack_ecache.h
@@ -65,14 +65,9 @@ struct nf_ct_event_notifier {
 	int (*exp_event)(unsigned int events, const struct nf_exp_event *item);
 };
 
-#ifdef CONFIG_NF_CONNTRACK_CHAIN_EVENTS
-extern int nf_conntrack_register_notifier(struct net *net, struct notifier_block *nb);
-extern int nf_conntrack_unregister_notifier(struct net *net, struct notifier_block *nb);
-#else
-int nf_conntrack_register_notifier(struct net *net,
+void nf_conntrack_register_notifier(struct net *net,
 				   const struct nf_ct_event_notifier *nb);
 void nf_conntrack_unregister_notifier(struct net *net);
-#endif
 
 void nf_ct_deliver_cached_events(struct nf_conn *ct);
 int nf_conntrack_eventmask_report(unsigned int eventmask, struct nf_conn *ct,

chenmozhijin, 2024-09-13:
看看有没有这个文件`/usr/share/firewall4/templates/zone-fullcone.uc`

jsdfhasuh, 2024-09-13:
> 看看有没有这个文件`/usr/share/firewall4/templates/zone-fullcone.uc`

这个文件有

jsdfhasuh, 2024-09-13:
nft_fullcone.ko这个文件没有

chenmozhijin, 2024-09-13:
应该是你换的这个内核里没有这个内核模块
https://github.com/fullcone-nat-nftables/nft-fullcone

jsdfhasuh, 2024-09-13:
> 应该是你换的这个内核里没有这个内核模块 https://github.com/fullcone-nat-nftables/nft-fullcone

这个怎么安装上？

jsdfhasuh, 2024-09-14:
已经安装上这个模块，但是luci界面还是显示没启用
[image]
[image]
[image]

jsdfhasuh, 2024-09-14:
已经解决，谢谢大佬

jsdfhasuh, 2024-09-14:
已经编译对应的模块

closed 2024-09-14 by jsdfhasuh