chenmozhijin/turboacc #30: Can't build for OpenWRT 23.05.5

upanasiuk, 2024-09-27:
I can't build Turboacc with SFE enabled for [OpenWRT ](https://github.com/openwrt/openwrt) [v23.05.5](https://github.com/openwrt/openwrt/tree/v23.05.5).

```
/home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/fast-classifier/fast-classifier.c:39:10: fatal error: sfe_backport.h: No such file or directory
   39 | #include <sfe_backport.h>
      |          ^~~~~~~~~~~~~~~~
```
Builds and works fine with `--no-sfe`.

diffconfig:
```
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_tplink_archer-c6-v2=y
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_AUTOREMOVE=y
CONFIG_BPF_TOOLCHAIN_BUILD_LLVM=y
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_BUILDBOT=y
CONFIG_COLLECT_KERNEL_DEBUG=y
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_IB=y
CONFIG_IMAGEOPT=y
CONFIG_JSON_CYCLONEDX_SBOM=y
CONFIG_KERNEL_BUILD_DOMAIN="buildhost"
CONFIG_KERNEL_BUILD_USER="builder"
# CONFIG_KERNEL_KALLSYMS is not set
CONFIG_MAKE_TOOLCHAIN=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_adb=m
CONFIG_PACKAGE_adb-enablemodem=m
CONFIG_PACKAGE_ath10k-board-qca9887=m
CONFIG_PACKAGE_ath10k-board-qca9888=m
CONFIG_PACKAGE_ath10k-board-qca988x=m
CONFIG_PACKAGE_ath10k-board-qca9984=m
CONFIG_PACKAGE_ath10k-board-qca99x0=m
CONFIG_PACKAGE_ath10k-firmware-qca9887-ct=m
CONFIG_PACKAGE_ath10k-firmware-qca9887-ct-full-htt=m
CONFIG_PACKAGE_ath10k-firmware-qca9888-ct=m
CONFIG_PACKAGE_ath10k-firmware-qca988x-ct=m
CONFIG_PACKAGE_ath10k-firmware-qca9984-ct=m
CONFIG_PACKAGE_ath10k-firmware-qca99x0-ct=m
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_e2fsprogs=m
CONFIG_PACKAGE_fconfig=m
CONFIG_PACKAGE_fritz-tffs=m
CONFIG_PACKAGE_kmod-ath10k-ct=m
CONFIG_PACKAGE_kmod-hwmon-core=m
CONFIG_PACKAGE_kmod-nft-fullcone=y
CONFIG_PACKAGE_kmod-rtc-ds1307=m
CONFIG_PACKAGE_kmod-rtc-ds1374=m
CONFIG_PACKAGE_kmod-rtc-pcf8563=m
CONFIG_PACKAGE_kmod-tcp-bbr=y
CONFIG_PACKAGE_libblkid=m
CONFIG_PACKAGE_libbpf=m
[... 137 lines cut ...]
make[3]: Leaving directory '/home/builder/openwrt/package/turboacc/shortcut-fe/shortcut-fe'
time: package/turboacc/shortcut-fe/shortcut-fe/compile#0.74#0.03#0.76
make[3]: Entering directory '/home/builder/openwrt/package/turboacc/shortcut-fe/fast-classifier'
rm -f /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/fast-classifier/.built
touch /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/fast-classifier/.built_check
make -j1 -C "/home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.15.167" KCFLAGS="-ffile-prefix-map=/home/builder/openwrt/build_dir/target-mips_24kc_musl=target-mips_24kc_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/builder/openwrt/staging_dir/host/include -I/home/builder/openwrt/staging_dir/hostpkg/include -I/home/builder/openwrt/staging_dir/target-mips_24kc_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mips-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="builder" KBUILD_BUILD_HOST="buildhost" KBUILD_BUILD_TIMESTAMP="Mon Sep 23 12:34:46 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/builder/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/ath10k-ct.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/batman-adv.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/button-hotplug.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/cryptodev-linux.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/dahdi-linux.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/dmx_usb_module.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/gpio-button-hotplug.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/jool.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/mac80211.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/macremapper.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/mdio-netlink.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/mt76.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/mtd-rw.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/nat46.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/netatop.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/nft-fullcone.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/openvswitch.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/ovpn-dco.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/r8101.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/r8125.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/r8126.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/r8168.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/rtl8812au-ct.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/rtpengine.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/shortcut-fe.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/siit.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/trelay.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/ubootenv-nvram.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/usb-serial-xr_usb_serial_common.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/v4l2loopback.symvers /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/xtables-addons.symvers" KERNELRELEASE=5.15.167  M="/home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/fast-classifier" CONFIG_FAST_CLASSIFIER=m EXTRA_CFLAGS+="-DSFE_SUPPORT_IPV6" modules
make[4]: Entering directory '/home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.15.167'
  CC [M]  /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/fast-classifier/fast-classifier.o
/home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/fast-classifier/fast-classifier.c:39:10: fatal error: sfe_backport.h: No such file or directory
   39 | #include <sfe_backport.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [scripts/Makefile.build:289: /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/fast-classifier/fast-classifier.o] Error 1
make[4]: *** [Makefile:1907: /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/fast-classifier] Error 2
make[4]: Leaving directory '/home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.15.167'
make[3]: *** [Makefile:107: /home/builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/fast-classifier/.built] Error 2
make[3]: Leaving directory '/home/builder/openwrt/package/turboacc/shortcut-fe/fast-classifier'
time: package/turboacc/shortcut-fe/fast-classifier/compile#1.51#0.16#1.60
    ERROR: package/turboacc/shortcut-fe/fast-classifier failed to build.
make[2]: *** [package/Makefile:129: package/turboacc/shortcut-fe/fast-classifier/compile] Error 1
make[2]: Leaving directory '/home/builder/openwrt'
make[1]: *** [package/Makefile:123: /home/builder/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/openwrt'
make: *** [/home/builder/openwrt/include/toplevel.mk:233: world] Error 2
```

chenmozhijin, 2024-09-28:
I did a [compilation test](https://github.com/chenmozhijin/turboacc/actions/runs/11073481324) and there was no problem

I think the problem you are experiencing is related to your configuration. You have CONFIG_BUILDBOT enabled which will compile all kernel modules, including kmod-fast-classifier and kmod-shortcut-fe-cm. These modules conflict with each other. And there may be compatibility issues with other plug-ins. If you need Shortcut-FE, you need to set
```
CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_SHORTCUT_FE=y
```
and you may need to disable CONFIG_BUILDBOT

I haven't actually used Shortcut-FE so I don't know much about it. It is said that it can improve the performance of small packet forwarding, but it will also increase instability and cause some strange problems

upanasiuk, 2024-09-29:
You're right.
My problem was in my configuration. I changed it (CONFIG_BUILDBOT is disabled and much more changed) and the build was successful with SFE enabled. It works 👌 As for now, my router can transmit 1 Gbps through NAT using insignificant CPU load (using Shortcut-FE).

closed 2024-09-29 by upanasiuk