chenmozhijin/turboacc #34: 最新的firewall4,patch文件有效性存疑。

NextAbyss, 2024-12-24:
[image]
我下载了最新的firewall4的包，如图内我搜索ruleset.uc文件的 redirect in fw4.redirects(`dstnat_${zone.name}`)): %}，提示搜不到，所以好奇当前的patch文件是否已经失效？

chenmozhijin, 2024-12-24:
你看看有没有`usr/share/firewall4/templates/zone-fullcone.uc`文件,patch前是没有的

我解firewall4_2024.12.18~18fc0ead-r1_x86_64.ipk的包在ruleset.uc中搜索
```
redirect in fw4.redirects(`dstnat_${zone.name}`)): %}
```
[image]

还有这其实是原文部分，不打补丁也有的

NextAbyss, 2024-12-25:
[image]
没有zone-fullcone.uc
我知道那是原文里的，emm那一行我搜到了,闹了个小乌龙，但是下一行我又有个疑问
[image]
画箭头的地方   patch里显示这一行不是需要修改的地方，但是和原文件里的不一样

chenmozhijin, 2024-12-28:
确实，可能是firewall4更新了，但这不影响patch
你说你“下载了最新的firewall4的包”你把它放在了哪里呢？
如果不是替换的`./package/network/config/firewall4`下的文件，则可能编译的是原来的firewall4

使用使用方法中的命令应该不会出现这样的问题

我进行测试的工作流运行:
https://github.com/chenmozhijin/turboacc/actions/runs/12484644668

你可以在[packages-master-FLOW_OFFLOAD](https://github.com/chenmozhijin/turboacc/actions/runs/12484644668/artifacts/2359918547)中找到patch过的最新的firewall4的apk包  
在[packages-openwrt-24.10-FLOW_OFFLOAD](https://github.com/chenmozhijin/turboacc/actions/runs/12484644668/artifacts/2359914608)中找到patch过的最新的firewall4的ipk包

closed 2025-01-21 by chenmozhijin