chenmozhijin/turboacc #35: 請教，為什麼我的加速似乎沒用？

linexu, 2025-01-02:
如圖，這個月，使用最新官方版本編譯，但是**無論怎麼勾選，“保存並應用”後，上面都提示運行中。**
編譯前使用的是帶SFE的命令安裝的。
請問這是什麼原因，應該怎麼解決？

[image]

linexu, 2025-01-03:
現在想使用不帶SFE重新編譯試試，但是在輸入
./scripts/feeds install -a
make menuconfig
之後，都有警告提示：
**WARNING: Makefile 'package/turboacc/luci-app-turboacc/Makefile' has a dependency on 'kmod-fast-classifier', which does not exist
WARNING: Makefile 'package/turboacc/luci-app-turboacc/Makefile' has a dependency on 'kmod-shortcut-fe-cm', which does not exist
WARNING: Makefile 'package/turboacc/luci-app-turboacc/Makefile' has a dependency on 'kmod-shortcut-fe-drv', which does not exist**

chenmozhijin, 2025-01-04:
警告是正常的

試一下用指令能不能關掉
開
```sh
uci set firewall.@defaults[0].flow_offloading="1"
uci set firewall.@defaults[0].flow_offloading_hw="1"
uci set firewall.@defaults[0].fullcone="1"
uci commit firewall
sysctl -w net.ipv4.tcp_congestion_control="bbr"
/etc/init.d/dnsmasq restart
/etc/init.d/firewall restart
```
關
```sh
uci set firewall.@defaults[0].flow_offloading="0"
uci set firewall.@defaults[0].flow_offloading_hw="0"
uci set firewall.@defaults[0].fullcone="0"
uci commit firewall
sysctl -w net.ipv4.tcp_congestion_control="cubic"
/etc/init.d/dnsmasq restart
/etc/init.d/firewall restart
```
在openwrt的終端執行看看有沒有效

linexu, 2025-01-06:
別提了，說起來就生氣。3天前準備使用不帶SFE重新編譯，目前已失敗了十多次，并且找不到原因，總在passwall的trojan-plus報錯，好像説是在server.o文件時失敗。但是奇怪的是，在内核6.6.68一直能編譯成功，目前下載的是内核6.6.69。除了這個差別外，我找不到其他的差異。現在我不安裝trojan-plus再試試。機器正在跑。。。

linexu, 2025-01-06:
> 警告是正常的
> 
> 試一下用指令能不能關掉 開
> 
> ```shell
> uci set firewall.@defaults[0].flow_offloading="1"
> uci set firewall.@defaults[0].flow_offloading_hw="1"
> uci set firewall.@defaults[0].fullcone="1"
> uci commit firewall
> sysctl -w net.ipv4.tcp_congestion_control="bbr"
> /etc/init.d/dnsmasq restart
> /etc/init.d/firewall restart
> ```
> 
> 關
> 
> ```shell
> uci set firewall.@defaults[0].flow_offloading="0"
> uci set firewall.@defaults[0].flow_offloading_hw="0"
> uci set firewall.@defaults[0].fullcone="0"
> uci commit firewall
> sysctl -w net.ipv4.tcp_congestion_control="cubic"
> /etc/init.d/dnsmasq restart
> /etc/init.d/firewall restart
> ```
> 
> 在openwrt的終端執行看看有沒有效

【関】
root@OpenWrt:~# uci set firewall.@defaults[0].flow_offloading="0"
root@OpenWrt:~# uci set firewall.@defaults[0].flow_offloading_hw="0"
root@OpenWrt:~# uci set firewall.@defaults[0].fullcone="0"
root@OpenWrt:~# uci commit firewall
root@OpenWrt:~# sysctl -w net.ipv4.tcp_congestion_control="cubic"
net.ipv4.tcp_congestion_control = cubic
root@OpenWrt:~# /etc/init.d/dnsmasq restart
udhcpc: started, v1.37.0
udhcpc: broadcasting discover
udhcpc: broadcasting select for 192.168.1.200, server 192.168.1.254
udhcpc: lease of 192.168.1.200 obtained from 192.168.1.254, lease time 4294967295
root@OpenWrt:~# /etc/init.d/firewall restart
Section @zone[0] (lan) fullcone in defaults not enabled, ignore zone fullcone settings
Section @zone[1] (wan) fullcone in defaults not enabled, ignore zone fullcone settings
Section parentcontrol option 'reload' is not supported by fw4
Section homeproxy_pre specifies unreachable path '/var/run/homeproxy/fw4_pre.nft', ignoring section
Section homeproxy_post specifies unreachable path '/var/run/homeproxy/fw4_post.nft', ignoring section
Section mia option 'reload' is not supported by fw4
Section openclash specifies unreachable path '/var/etc/openclash.include', ignoring section
Section passwall option 'reload' is not supported by fw4
Section passwall specifies unreachable path '/var/etc/passwall.include', ignoring section
Section passwall_server option 'reload' is not supported by fw4
Section passwall_server specifies unreachable path '/var/etc/passwall_server.include', ignoring section
Section shadowsocksr option 'reload' is not supported by fw4
Automatically including '/usr/share/nftables.d/table-post/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/dstnat/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/forward/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/srcnat/20-miniupnpd.nft'
sh: /etc/parentcontrol.include: line 1: /etc/init.d/parentcontrol: Permission denied
Include '/etc/parentcontrol.include' failed with exit code 126
sh: /etc/mia.include: line 1: /etc/init.d/mia: Permission denied
Include '/etc/mia.include' failed with exit code 126

[image]

【開】
root@OpenWrt:~# uci set firewall.@defaults[0].flow_offloading="1"
root@OpenWrt:~# uci set firewall.@defaults[0].flow_offloading_hw="1"
root@OpenWrt:~# uci set firewall.@defaults[0].fullcone="1"
root@OpenWrt:~# uci commit firewall
root@OpenWrt:~# sysctl -w net.ipv4.tcp_congestion_control="bbr"
net.ipv4.tcp_congestion_control = bbr
root@OpenWrt:~# /etc/init.d/dnsmasq restart
udhcpc: started, v1.37.0
udhcpc: broadcasting discover
udhcpc: broadcasting select for 192.168.1.200, server 192.168.1.254
udhcpc: lease of 192.168.1.200 obtained from 192.168.1.254, lease time 4294967295
root@OpenWrt:~# /etc/init.d/firewall restart
Section @zone[1] (wan) IPv4 fullcone enabled for zone 'wan'
Section parentcontrol option 'reload' is not supported by fw4
Section homeproxy_pre specifies unreachable path '/var/run/homeproxy/fw4_pre.nft', ignoring section
Section homeproxy_post specifies unreachable path '/var/run/homeproxy/fw4_post.nft', ignoring section
Section mia option 'reload' is not supported by fw4
Section openclash specifies unreachable path '/var/etc/openclash.include', ignoring section
Section passwall option 'reload' is not supported by fw4
Section passwall specifies unreachable path '/var/etc/passwall.include', ignoring section
Section passwall_server option 'reload' is not supported by fw4
Section passwall_server specifies unreachable path '/var/etc/passwall_server.include', ignoring section
Section shadowsocksr option 'reload' is not supported by fw4
Automatically including '/usr/share/nftables.d/table-post/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/dstnat/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/forward/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/srcnat/20-miniupnpd.nft'
sh: /etc/parentcontrol.include: line 1: /etc/init.d/parentcontrol: Permission denied
Include '/etc/parentcontrol.include' failed with exit code 126
sh: /etc/mia.include: line 1: /etc/init.d/mia: Permission denied
Include '/etc/mia.include' failed with exit code 126

[image]

使用這些命令是有用的，但是涉及到另外2個插件，說沒有權限。
另外建議您修復時，順便完善一下這個：
[image]

再請教個問題，其他人使用您的插件，但是似乎他們修改了關於“IPv6 全锥形 NAT”的解釋。
您似乎不建議開啓，但他們建議開啓？？？
[image]

linexu, 2025-01-06:
> 別提了，說起來就生氣。3天前準備使用不帶SFE重新編譯，目前已失敗了十多次，并且找不到原因，總在passwall的trojan-plus報錯，好像説是在server.o文件時失敗。但是奇怪的是，在内核6.6.68一直能編譯成功，目前下載的是内核6.6.69。除了這個差別外，我找不到其他的差異。現在我不安裝trojan-plus再試試。機器正在跑。。。

這次沒加trojan-plus就順利通過了，編譯成功。

linexu, 2025-01-06:
> > 別提了，說起來就生氣。3天前準備使用不帶SFE重新編譯，目前已失敗了十多次，并且找不到原因，總在passwall的trojan-plus報錯，好像説是在server.o文件時失敗。但是奇怪的是，在内核6.6.68一直能編譯成功，目前下載的是内核6.6.69。除了這個差別外，我找不到其他的差異。現在我不安裝trojan-plus再試試。機器正在跑。。。
> 
> 這次沒加trojan-plus就順利通過了，編譯成功。

這次編譯的是不帶SFE的，可是依舊不行，無論關閉、開啓，都是顯示綠色“運行中”。

chenmozhijin, 2025-02-07:
867d56398bfc8c07c959d26a905ecd102fedc394 已修復

closed 2025-02-07 by chenmozhijin