chenmozhijin/turboacc #37: 执行make menuconfig出现的提示，以及测试使用时勾选BBR和流量分载，无论如何都是运行中

asdchange7, 2025-01-31:
tmp/.config-package.in:62660:error: recursive dependency detected!
tmp/.config-package.in:62660:	symbol PACKAGE_luci-app-turboacc depends on PACKAGE_kmod-nft-fullcone
tmp/.config-package.in:16468:	symbol PACKAGE_kmod-nft-fullcone is selected by PACKAGE_luci-app-turboacc
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

chenmozhijin, 2025-02-07:
[image]

[image]

第一个问题我复现不了
第二个问题已修复

asdchange7, 2025-02-08:
> [image]
> 
> [image]
> 
> 第一个问题我复现不了 第二个问题已修复

我是用ubuntu24.04编译的immortalwrt，也许是因为这个？

chenmozhijin, 2025-02-09:
删除http://package/network/utils/fullconenat-nft/试试，immortalwrt自带一个[nft-fullcone](https://github.com/immortalwrt/immortalwrt/blob/master/package/network/utils/fullconenat-nft/Makefile)可能是它导致的