chenmozhijin/turboacc #38: 貌似与24.10防火墙的offload冲突

gitbruc, 2025-02-04:
[image]

[image]

1.当开启防火墙的offload时候，turboacc无论开flow还是fe都会显示已开启flow
2.当关闭防火墙的offload时候，turboacc无论开flow还是fe都会显示已关闭

这样flow是可以正常使用的，但是fe无法使用

~~虽然我一直flow，但我还是提一下吧~~

chenmozhijin, 2025-02-04:
没问题turboacc只是开关了一个防火墙开关，两个效果是一样的

https://github.com/chenmozhijin/turboacc/blob/142ed03e2468dc75ca43f960311be9128a142b21/luci-app-turboacc/root/etc/init.d/turboacc#L97-L97

gitbruc, 2025-02-04:
对,我意思是sfe无法启动了

gitbruc, 2025-02-04:
不对我说错了，不是没法用，是切换之后无法正常显示

[image]

[image]

gitbruc, 2025-02-04:
必须重启系统才能显示

chenmozhijin, 2025-02-07:
867d56398bfc8c07c959d26a905ecd102fedc394 已修復

gitbruc, 2025-02-08:
感谢大佬🙏🏻️