chenmozhijin/turboacc #42: Failed to compile with the latest Main branch 6.12 kernel.

chenrui9119, 2025-05-07:
With or without --no-sfe either.

```
bash add_turboacc.sh --no-sfe
kernel version: 6.6 6.12, No SFE: true
./package/turboacc already exists, delete it? [Y/N]
Y
]正克隆到 '/tmp/tmp.DkKmEc3Vjy/turboacc/nft-fullcone'...
remote: Enumerating objects: 15, done.
remote: Counting objects: 100% (15/15), done.
remote: Compressing objects: 100% (15/15), done.
remote: Total 15 (delta 0), reused 11 (delta 0), pack-reused 0 (from 0)
接收对象中: 100% (15/15), 16.08 KiB | 2.01 MiB/s, 完成.
正克隆到 '/tmp/tmp.DkKmEc3Vjy/turboacc/turboacc'...
remote: Enumerating objects: 40, done.
remote: Counting objects: 100% (40/40), done.
remote: Compressing objects: 100% (26/26), done.
remote: Total 40 (delta 0), reused 29 (delta 0), pack-reused 0 (from 0)
接收对象中: 100% (40/40), 238.35 KiB | 4.77 MiB/s, 完成.
正克隆到 '/tmp/tmp.DkKmEc3Vjy/package'...
remote: Enumerating objects: 144, done.
remote: Counting objects: 100% (144/144), done.
remote: Compressing objects: 100% (110/110), done.
remote: Total 144 (delta 39), reused 109 (delta 13), pack-reused 0 (from 0)
接收对象中: 100% (144/144), 121.47 KiB | 3.37 MiB/s, 完成.
处理 delta 中: 100% (39/39), 完成.
./target/linux/generic/hack-6.6/952-add-net-conntrack-events-support-multiple-registrant.patch already exists, delete.
cp: 对 '/tmp/tmp.DkKmEc3Vjy/package/hack-6.12/952-add-net-conntrack-events-support-multiple-registrant.patch' 调用 stat 失败: 没有那个文件或目录
Finish

```
Then I try copy 952 to hack-6.12 then make, and it will fail like this:

```
make[6]: *** [scripts/Makefile.build:229: /home/parallels/openwrt/build_dir/target-x86_64_musl/linux-x86_64/nft-fullcone/nft_ext_fullcone.o] Error 1
make[5]: *** [/home/parallels/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.26/Makefile:1949: /home/parallels/openwrt/build_dir/target-x86_64_musl/linux-x86_64/nft-fullcone] Error 2
make[4]: *** [Makefile:224: __sub-make] Error 2
make[4]: Leaving directory '/home/parallels/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.26'
make[3]: *** [Makefile:48: /home/parallels/openwrt/build_dir/target-x86_64_musl/linux-x86_64/nft-fullcone/.built] Error 2
make[3]: Leaving directory '/home/parallels/openwrt/package/turboacc/nft-fullcone'
time: package/turboacc/nft-fullcone/compile#1.37#0.24#1.60
    ERROR: package/turboacc/nft-fullcone failed to build.
make[2]: *** [package/Makefile:189: package/turboacc/nft-fullcone/compile] Error 1

```

xiaouex, 2025-05-10:
Same here,and it cannot found some patches when execute curl -sSL https://raw.githubusercontent.com/chenmozhijin/turboacc/luci/add_turboacc.sh -o add_turboacc.sh && bash add_turboacc.sh

catsimple, 2025-05-13:
```
make -j1 -C /home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/linux-6.12.28 KCFLAGS="-fmacro-prefix-map=/home/catsimple/openwrt/build_dir/target-x86_64_glibc=target-x86_64_glibc -fno-caller-saves " HOSTCFLAGS="-I/home/catsimple/openwrt/staging_dir/host/include -I/home/catsimple/openwrt/staging_dir/hostpkg/include -I/home/catsimple/openwrt/staging_dir/target-x86_64_glibc/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-gnu-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon May 12 16:10:03 2025" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/catsimple/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/symvers/button-hotplug.symvers /home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/symvers/cryptodev-linux.symvers /home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/symvers/mac80211.symvers" CC="x86_64-openwrt-linux-gnu-gcc" KERNELRELEASE=6.12.28 M="/home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/nft-fullcone" EXTRA_CFLAGS=""  modules
make[4]: Entering directory '/home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/linux-6.12.28'
  CC [M]  /home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/nft-fullcone/nft_ext_fullcone.o
/home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/nft-fullcone/nft_ext_fullcone.c:280:21: error: initialization of 'int (*)(const struct nft_ctx *, const struct nft_expr *)' from incompatible pointer type 'int (*)(const struct nft_ctx *, const struct nft_expr *, const struct nft_data **)' [-Werror=incompatible-pointer-types]
  280 |         .validate = nft_fullcone_validate,
      |                     ^~~~~~~~~~~~~~~~~~~~~
/home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/nft-fullcone/nft_ext_fullcone.c:280:21: note: (near initialization for 'nft_fullcone_ipv4_ops.validate')
/home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/nft-fullcone/nft_ext_fullcone.c:320:21: error: initialization of 'int (*)(const struct nft_ctx *, const struct nft_expr *)' from incompatible pointer type 'int (*)(const struct nft_ctx *, const struct nft_expr *, const struct nft_data **)' [-Werror=incompatible-pointer-types]
  320 |         .validate = nft_fullcone_validate,
      |                     ^~~~~~~~~~~~~~~~~~~~~
/home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/nft-fullcone/nft_ext_fullcone.c:320:21: note: (near initialization for 'nft_fullcone_ipv6_ops.validate')
/home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/nft-fullcone/nft_ext_fullcone.c:379:21: error: initialization of 'int (*)(const struct nft_ctx *, const struct nft_expr *)' from incompatible pointer type 'int (*)(const struct nft_ctx *, const struct nft_expr *, const struct nft_data **)' [-Werror=incompatible-pointer-types]
  379 |         .validate = nft_fullcone_validate,
      |                     ^~~~~~~~~~~~~~~~~~~~~
/home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/nft-fullcone/nft_ext_fullcone.c:379:21: note: (near initialization for 'nft_fullcone_inet_ops.validate')
cc1: all warnings being treated as errors
make[6]: *** [scripts/Makefile.build:229: /home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/nft-fullcone/nft_ext_fullcone.o] Error 1
make[5]: *** [/home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/linux-6.12.28/Makefile:1949: /home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/nft-fullcone] Error 2
make[4]: *** [Makefile:224: __sub-make] Error 2
make[4]: Leaving directory '/home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/linux-6.12.28'
make[3]: *** [Makefile:48: /home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/nft-fullcone/.built] Error 2
make[3]: Leaving directory '/home/catsimple/openwrt/package/turboacc/nft-fullcone'
time: package/turboacc/nft-fullcone/compile#0.52#0.53#1.17
    ERROR: package/turboacc/nft-fullcone failed to build.
make[2]: *** [package/Makefile:189: package/turboacc/nft-fullcone/compile] Error 1
make[2]: Leaving directory '/home/catsimple/openwrt'
make[1]: *** [package/Makefile:183: /home/catsimple/openwrt/staging_dir/target-x86_64_glibc/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/catsimple/openwrt'
make: *** [/home/catsimple/openwrt/include/toplevel.mk:233：world] 错误 2

```

xiaouex, 2025-05-13:
> ```
> make -j1 -C /home/catsimple/openwrt/build_dir/target-x86_64_glibc/linux-x86_64/linux-6.12.28 KCFLAGS="-fmacro-prefix-map=/home/catsimple/openwrt/build_dir/target-x86_64_glibc=target-x86_64_glibc [...]

You can pick the commit from immortalwrt，[7d85a71fffae761477fa4a4178af42df85e06455](https://github.com/immortalwrt/immortalwrt/commit/7d85a71fffae761477fa4a4178af42df85e06455),to fix build with kernel 6.12

chenmozhijin, 2025-05-14:
This issue should have been fixed and should compile fine now
https://github.com/chenmozhijin/turboacc/actions/runs/15001293183

miccjing, 2025-05-22:
cp: cannot stat '/tmp/tmp.Uo7YBkryKA/package/hack-6.12/952-add-net-conntrack-events-support-multiple-registrant.patch': No such file or directory
Finish

shjy223534, 2025-05-23:
> 该问题应该已经修复，现在可以正常编译了 https://github.com/chenmozhijin/turboacc/actions/runs/15001293183

还是不行啊，imm主要分支执行您的那个命令后还是报错：cp: 对 '/tmp/tmp.WOepeEJMie/package/hack-6.12/952-add-net-conntrack-events-support-multiple-registrant.patch' 调用 stat 失败: 没有那个文件或目录

chenmozhijin, 2025-05-24:
现在呢？

shjy223534, 2025-05-26:
> 现在呢？

可以了，谢谢，麻烦了

closed 2025-05-31 by chenmozhijin